PX4/PX4-Autopilot #22720: [Bug] px4_mtd: n_partitions_current could be 0

w2016561536, 2024-02-05:
### Describe the bug

Hello, I am trying to port PX4 to a new board and found a problem in code.  
In [https://github.com/PX4/PX4-Autopilot/blob/c78389a855391adaf2dd05f58afcbc6b1500ff1a/platforms/nuttx/src/px4/common/px4_mtd.cpp#L203](url), the varible ` instance->n_partitions_current ` could be 0. And it triggers mcu divided by 0 assertion. Moreover, nblocks do not be used. 

### To Reproduce

N/A

### Expected behavior

N/A

### Flight Log

N/A

### Software Version

N/A

### Flight controller

N/A

### Vehicle type

None

yuxuanwu1234567, 2024-02-05:
Borrowing the building, may I ask why the drone can take off in altitude mode, but in self stabilization mode, the command is temporarily rejected

w2016561536, 2024-02-05:
> Borrowing the building, may I ask why the drone can take off in altitude mode, but in self stabilization mode, the command is temporarily rejected

Probably that your drone does not have a valid height sensor( baro, gps, lidar or others). BTW, please open an new issue if you have new question.