PX4/PX4-Autopilot #22728: [Bug] ninja: error: unknown target 'gz_x500'

pjanuszewski, 2024-02-06:
### Describe the bug

I'm trying to run the basic simulation with gazebo using the `make px4_sitl gz_x500` command. I followed the steps in documentation, rebooted computer and ran `make distclean` command, before trying to start simulation. I'm getting error: 
`ninja: error: unknown target 'gz_x500'`. It's worth noting that when I tried to run `make px4_sitl jmavsim`, the simulation works fine. I only have a problem with starting gazebo.

### Full output:
~/Git_repos/PX4-Autopilot$ make px4_sitl gz_x500
-- PX4 version: v1.15.0-alpha1-512-g17d55dddd6 (1.15.0)
-- Found PythonInterp: /home/user/miniconda3/bin/python3 (found suitable version "3.11.5", minimum required is "3") 
-- PX4 config file: /home/user/Git_repos/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 13.2.0
-- The C compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/miniconda3/bin/x86_64-conda-linux-gnu-cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/user/miniconda3/bin/x86_64-conda-linux-gnu-c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/user/miniconda3/bin/x86_64-conda-linux-gnu-cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: Release
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Found Java: /usr/bin/java (found version "11.0.21") 
-- ROMFS: ROMFS/px4fmu_common
Architecture:  amd64
==> CPACK_INSTALL_PREFIX = @DEB_INSTALL_PREFIX@
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/Git_repos/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1

### To Reproduce

1. `git clone https://github.com/PX4/PX4-Autopilot.git --recursive`
2. `bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`
3. Restarting the computer on completion
4. `make distclean`
5. `make px4_sitl gz_x500`

### Expected behavior

N/A

### Flight Log

N/A

### Software Version

I'm using the [last version of px4](https://github.com/PX4/PX4-Autopilot/tree/main)
PX4 version: v1.15.0
OS: Ubuntu 22.04

### Flight controller

N/A

### Vehicle type

None

vacoff, 2024-02-15:
I am not sure I didn't try to reproduce the error, but I guess the px4 comes with only gazebo classic. You should download gazebo garden or harmonic separately. That will uninstall the gazebo classic and then you can run your command with no issues.

frede791, 2024-02-15:
Also worth checking that the gz submodule has actually been initialized. These can sometimes be a bit flaky. 
`ls PX4-Autopilot/Tools/simulation/gz` If this folder is emtpy, the submodule has not been initialized correctly and you can fix this by running the following command from PX4 root:
`git submodule update --init --recursive`

pjanuszewski, 2024-02-18:
I've solved the issue. The compiler that was being used with the simulation was actually ```x86_64-conda-linux-gnu-g++``` from miniconda3 package. When I set the Ubuntu gcc 11.4.0 as the default compiler with
```
export CC=gcc
export CXX=g++
```
both gazebo and jmavsim work properly. I was also getting weird errors with dangling pointers before, while using miniconda compiler, but now everything works fine. Thanks for the advice.

closed 2024-02-18 by pjanuszewski

skol101, 2024-02-20:
> I've solved the issue. The compiler that was being used with the simulation was actually `x86_64-conda-linux-gnu-g++` from miniconda3 package. When I set the Ubuntu gcc 11.4.0 as the default [...]
Had the same issue. I though px4 was perfectly compatible with a conda env, but apparently, it's not.