PX4/PX4-Autopilot #22763: [Bug] IMU timing in UXRCE (ROS2) is inconsistent

Ubledeli, 2024-02-15:
### Describe the bug

IMU reading should be around 200Hz (time between two consecutive messages should be 5ms)
It is mostly around 5ms but it sometimes jumps to 10ms or even more
That makes the average reading rate around 180Hz on v1.14 and around 100Hz on the main branch

uorb top in Mavlink console reports the same as ROS2

Is there a way to set this message rate in the source or via parameters

Current setup is pixhawk6x connected to a PC over ethernet

### To Reproduce

connect FCU to the PC over ethernet and start UXRCE agent
micro-xrce-dds-agent udp4 -p 8888 -m dds
then, check the message rate either with:
ros2 topic hz /fmu/out/sensor_combined 
or
uorb top in Mavlink console

### Expected behavior

Messages should arrive uniformly with a time delay of 5ms between them
Ideally, there should be a way using a parameter or otherwise, to set this time (or frequency) 

### Screenshot / Media

[image]

### Flight Log

no log present

### Software Version

v1.14

### Flight controller

pixhawk 6x

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

ethernet communication is set following this [guide](https://docs.px4.io/main/en/middleware/uxrce_dds.html)

Edekheh, 2025-02-18:
The publication rate of sensor_combined is related to:

#define UXRCE_DEFAULT_POLL_RATE 10
(Defined in src/modules/uxrce_dds_topics.h.em)
Parameters:
IMU_INTEG_RATE
IMU_GYRO_RATEMAX
If I understand correctly, the uxrce publisher needs to run at least 2× faster than the sensor_combined publishing rate (which depends on the two parameters above). This follows the principle of the Nyquist frequency, as explained in [Wikipedia](https://en.wikipedia.org/wiki/Nyquist_frequency).

mrpollo, 2025-02-19:
Please don't resurrect older issues, it won't help you get help. Please create a new issue and try adding as much details as you can. Feel free to tag me.

closed 2025-02-19 by mrpollo