PX4/PX4-Autopilot #22767: Acceleration and Gyro sensors send STALE error messages

frede791, 2024-02-15:
### Describe the bug

When running gz simulation normally, two different errors appears shortly after starting: 

```shell
ERROR [sensors] Accel #0 fail:  STALE!
ERROR [sensors] Gyro #0 fail:  STALE!
```

To my knowledge, this was first reported by @slgrobotics in https://github.com/PX4/PX4-Autopilot/pull/22550#issuecomment-1946286865 but its root cause doesn't seem to lie in PX4 or Gazebo sim as the word STALE!  does not appear anywhere in either repository.

### To Reproduce

On current main branch (74303a79e195083144c9d63fd2346b5828015a31) run 

`make px4_sitl gz_x500` 

### Expected behavior

No error should appear. 

### Flight Log

N/A

### Software Version

Current main 74303a79e195083144c9d63fd2346b5828015a31

### Flight controller

SITL

### Vehicle type

None

frede791, 2024-02-15:
The error seems printed by this line of code related to the mavlink data validator:
https://github.com/PX4/PX4-Autopilot/blob/74303a79e195083144c9d63fd2346b5828015a31/src/modules/sensors/voted_sensors_update.cpp#L312C14-L312C79

PerFrivik, 2024-02-16:
@dagar This issue comes from your PR: [#22736](https://github.com/PX4/PX4-Autopilot/pull/22736)

@frede791 is taking a look at why this happens. 

There is also some other warning, telling us that there is no gyro selected, but that has been around for a while. 

WARN  [vehicle_angular_velocity] no gyro selected, using sensor_gyro:0 1310988

frede791, 2024-02-16:
The error seems to be caused by removing the line `param set-default SENS_IMU_MODE 0` from the rcS file. It seems that when running gz it no longer gets set to any value leading to the `checkFailover` function being called in the `voted_sensors_update.cpp` file which creates this error.

goupil35000, 2024-02-19:
Hello,
Is there any solution for these STALE messages. I am exactly in the same situation.
Thanks

PerFrivik, 2024-02-19:
@goupil35000 a temporary solution until @dagar replies is to interactively rebase on your branch and remove the following commit: 

https://github.com/PX4/PX4-Autopilot/pull/22736

goupil35000, 2024-02-19:
@PerFrivik Many Thanks. 

I will wait until  @dagar's changes. I will retest after.

frede791, 2024-02-22:
@dagar @Jaeyoung-Lim @PerFrivik @slgrobotics As discussed yesterday in the call, the issue can be fixed by adding some noise distributions to the imu sensor in the sdf file. I test ran this using the screenshot below and it worked fine. @Jaeyoung-Lim mentioned that we should use the values based on what was used in gazebo-classic. I will look into this.

[image]

dagar, 2024-02-22:
https://github.com/PX4/PX4-gazebo-models/pull/33 merged

Jaeyoung-Lim, 2024-02-22:
@dagar @frede791 Let the submodule dance begin!

VBS2004, 2024-02-28:
Is it done?

frede791, 2024-03-05:
@VBS2004 @slgrobotics @PerFrivik @goupil35000 The gz submodule has been updated by PR https://github.com/PX4/PX4-Autopilot/pull/22841. Please check that the models you need are working without the error message.

closed 2024-03-05 by frede791

slgrobotics, 2024-03-06:
Confirming that the STALE messages are no longer there, when x500 or lawnmower are built from latest "_main_". Thanks!

PerFrivik, 2024-03-07:
@vibgyor-s This must have been idle for quite a while (20+ minutes)? 

I also get these issues.

VBS2004, 2024-03-07:
> @vibgyor-s It seems the system has been idle for a long time (20+ minutes). I also receive these warnings, which could be related to the estimator. With minimal acceleration data for fusion, the estimator may struggle to handle drift effectively.

How to fix this.I am also receiving the same errors when running mavsdk example of gotolocation ,it just rolls and crashes.

PerFrivik, 2024-03-07:
> How to fix this.I am also receiving the same errors when running mavsdk example of gotolocation ,it just rolls and crashes.

Do you also just crash, if you instantly take off and do the goto?

frede791, 2024-03-07:
@VBS2004 Maybe this warrants making a separate issue?

VBS2004, 2024-03-08:
> @VBS2004 Maybe this warrants making a separate issue?

Idk I am a complete beginner in this stuff Should I open a new issue?

frede791, 2024-03-08:
It would be good to know why this is occurring as it is unexpected behaviour.

frede791, 2024-03-11:
@PerFrivik Are you running the newest version of the rover?

slgrobotics, 2024-03-11:
I just did:
```
mkdir src0; cd src0
git clone https://github.com/PX4/PX4-Autopilot.git --single-branch --recursive
cd PX4-Autopilot/
make px4_sitl gz_r1_rover
```
 There are no STALE messages, but the following still are present:
```
Warning [Utils.cc:130] [/sdf/world[@name="default"]
/spherical_coordinates/altitude:.../worlds/default.sdf:L153]: XML Element[altitude], 
child of element[spherical_coordinates], not defined in SDF. 
...
pxh> 
pxh> WARN  [health_and_arming_checks] Preflight: GPS Vertical Pos Drift too high
WARN  [health_and_arming_checks] Preflight: GPS Vertical Pos Drift too high
WARN  [health_and_arming_checks] Preflight: GPS Horizontal Pos Drift too high
```
The "_altitude_" thing has been fixed (by "_elevation_") - but the submodule hash hasn't been brought to latest.

VBS2004, 2024-03-11:
The problems in gz_x500 is resolved
`trajectory_msg = TrajectorySetpoint()
            trajectory_msg.timestamp = int(Clock().now().nanoseconds / 1000)
            trajectory_msg.velocity[0] = float('nan')
            trajectory_msg.velocity[1] = float('nan')
            trajectory_msg.velocity[2] = float('nan')
            trajectory_msg.position[0] = 10
            trajectory_msg.position[1] = 10
            trajectory_msg.position[2] = -20
            trajectory_msg.acceleration[0] = float('nan')
            trajectory_msg.acceleration[1] = float('nan')
            trajectory_msg.acceleration[2] = float('nan')
            trajectory_msg.yaw = float('nan')
            trajectory_msg.yawspeed = self.yaw

            self.publisher_trajectory.publish(trajectory_msg)`
this code works without any issues except this time jump detected. Resetting time synchroniser.
But the example I took directly from mavsdk is not working it gives so many errors and drone just crashes as I have mentioned in this issue [https://github.com/PX4/PX4-Autopilot/issues/22861](url) 
Here's the errors:
`ERROR [health_and_arming_checks] angular velocity no longer valid (non-finite values)
INFO  [tone_alarm] battery warning (fast)
WARN  [health_and_arming_checks] Preflight Fail: ESC failure detected
WARN  [health_and_arming_checks] Preflight Fail: Flight termination active
INFO  [commander] Disarmed by lockdown	
WARN  [health_and_arming_checks] Preflight Fail: height estimate error
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
WARN  [health_and_arming_checks] Preflight Fail: position estimate error
WARN  [health_and_arming_checks] Preflight Fail: ESC failure detected
WARN  [health_and_arming_checks] Preflight Fail: Flight termination active
`

PerFrivik, 2024-03-11:
> @PerFrivik Are you running the newest version of the rover?

The one on main, or do I need to checkout a different commit hash?

slgrobotics, 2024-03-12:
@PerFrivik  - maybe, your "main" is not main enough. ;-)

Try:
```
mkdir src0; cd src0
git clone https://github.com/PX4/PX4-Autopilot.git --single-branch --recursive
cd PX4-Autopilot/
```
and then ```"meld <your PX4-Autopilot> <~/src0/PX4-Autopilot>"```

I noticed many times before that git doesn't properly update submodules every time, so I use _meld_ when in doubt.

PerFrivik, 2024-03-13:
I guess, my main was really not "main" enough hahah. I fixed it with a make distclean :+1:. Thanks for the help!

PerFrivik, 2024-03-13:
@slgrobotics I have never used meld before, what does that do?

slgrobotics, 2024-03-13:
@PerFrivik  -- https://en.wikipedia.org/wiki/Meld_(software)   very good Linux comparison tool

federicociresola, 2024-06-11:
I am having the problem with the main. I update all submodule and I have the last version of the main but when I run 
`make px4_sitl gz_x500` I  obtain the following error:
[image]

I thought that this error has been resolved