PX4/PX4-Autopilot #22774: [Fixedwings] Instantaneous path setpoint for vertical axis

Jaeyoung-Lim, 2024-02-19:
### Describe problem solved by the proposed feature

For fixed-wing vehicles, it is only possible to either specify a height setpoint or a height rate setpoint for TECS.

This is enough for most use cases in waypoint-based missions but is insufficient to guide the vehicle along a glide slope.

It would be very useful if one can guide the vehicle along a (feasible)glide slope for various use cases such as landing, navigation from a path planner.

### Describe your preferred solution

Currently, the longitudinal axis is completely decomposed. It would be great if we can specify the height as closest point on the glide path and the height rate, which would be the projected speed the vehicle is expected to descend along the path.could

### Describe possible alternatives

We could also introduce full 3D position setpoints and output acceleration setpoints to track the path. However, this might be less energy efficient compared to using TECS.

vacoff, 2024-02-20:
Would it also allow to use px4 with unpowered gliders?

Jaeyoung-Lim, 2024-02-20:
@vacoff You can already use px4 for unpowered gliders, if you want to stay on an arbitrary glide slope by regulating airspeed

vacoff, 2024-02-20:
What about gliders dropped from mr platforms with 0 airspeed at the start? I think px4 would have a hard time trying to level up the flight.

sfuhrer, 2024-02-20:
@KonradRudin @tstastny food for thought

Jaeyoung-Lim, 2024-02-20:
> What about gliders dropped from mr platforms with 0 airspeed at the start? I think px4 would have a hard time trying to level up the flight.

@vacoff I think this is not related to the feature we are discussing here. This would be about regulating airspeed using TECS, irrespective of being on a 3D path.

ryanjAA, 2024-02-29:
Great idea!

AshhadRasheed, 2024-09-02:
im a little late to this discussion, but i believe 1.10.1 and a few earlier releases had first-order and zero-order hold funtionality for height setpoints.

closed 2026-04-09 by github-actions[bot]