PX4/PX4-Autopilot #22784: attitude error about qd_red *= q

lixiaowei123, 2024-02-21:
### Describe problem solved by the proposed feature

https://flyingmachinearena.org/wp-content/publications/2018/breTCST18.pdf

Excuse me, I've been trying to figure out how this is derived about qd_red *= q, but I haven't found a good explanation yet.
[image]

### Describe your preferred solution

..

### Describe possible alternatives

..

### Additional context

..

bresch, 2024-02-21:
1. `qd_red(e_z, e_z_d)` is a quaternion that rotates the current "down" to the desired "down" vector.
2. The action `qd_red * q` is rotating the current full attitude such that the tilt is modified (i.e.: has the desired "down" vector) but the yaw is untouched.

Hope that helped.

lixiaowei123, 2024-02-22:
> 1. `qd_red(e_z, e_z_d)` is a quaternion that rotates the current "down" to the desired "down" vector.
> 2. The action `qd_red * q` is rotating the current full attitude such that the tilt is modified (i.e.: has the desired "down" vector) but the yaw is untouched.
> 
> Hope that helped.

Thank you very much for your recovery.The action qd_red * q  Why not write it as qd_red.inv * q

lixiaowei123, 2024-02-26:
> qd_red * q

Qd_red * q represents rotating the quaternion q corresponding to the current attitude according to the quaternion qd_red corresponding to the vertical error vector. The new quaternion obtained is the quaternion corresponding to the roll and pitch rotation of the drone. Does this mean？

bresch, 2024-02-27:
`Qd_red * q` gives a quaternion corresponding to the roll and pitch of `qd` but the yaw of `q`.

mengchaoheng, 2024-03-02:
Maybe some different from the paper, but it can prove that it is equivalent to the paper's formula

closed 2024-03-03 by bresch