PX4/PX4-Autopilot #22791: Orange Cube FW v1.14.0 Pitch command winds up while disarmed pre-flight and initial manual mode flight interval

autoflightlab, 2024-02-22:
### Describe the bug

Pitch command winds up during preflight while disarmed and continues to wind up in manual and does not reset to current pitch value when RTL engaged, leading to very large pitch upset , -80 deg nose down in less than 2 sec 

### To Reproduce

Extensive preflight checklist completed prior to flight, FW drone (plane) disarmed while waiting for takeoff approval,  Plane re-armed, takeff performed in manual mode, flew for about 90 sec prior to engaging RTL from RC Tx. Plane went nose down quickly, approximately 80 deg before recovered manually and flown to higher altitude whereupon RTL was engaged second time,  with no pitch transient.

### Expected behavior

Expected to see little or no pitch transient when switching  to RTL, not 80 deg nose down

### Screenshot / Media

[image]
Y axis is radians

### Flight Log

Your uploaded log file is available under:
https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Freview.px4.io%2Fplot_app%3Flog%3D1181fc5e-721b-4a0a-8c56-760c11c681b2&data=05%7C02%7Cmark.a.motter%40nasa.gov%7Cac54b712352c4bf8fb0708dc32f3570c%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C638441267919948797%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=FuC7cmLNmiMBEvHRU94lyZ%2Bg1I%2FA%2FZr1nQ35P6OzA1Q%3D&reserved=0

Description:
Feedback:
Vehicle type: Fixed Wing
Airframe: Generic Standard Plane, Standard Plane
Hardware: CUBEPILOT_CUBEORANGE
Vehicle UUID: 00060000000036323737323251090041002a
Software git hash: b8c541dd7277ed735139d7d1bfb829d61fbe29fb
Upload file name: log_163_2024-2-20-11-51-06.ulg

### Software Version

v 1.14.0

### Flight controller

Orange Cube

### Vehicle type

Fixed Wing

### How are the different components wired up (including port information)

Ailerons
Flaps
Elevator
Rudder 
Throttle

autoflightlab, 2024-03-01:
After looking at the data some more, the pitch command windup starts when armed and continued through manual mode.
Went back to V1.13.3, no issues with that version

bresch, 2024-03-18:
@sfuhrer Is that the same issue fixed by https://github.com/PX4/PX4-Autopilot/pull/22499 ?

sfuhrer, 2024-03-18:
Yes that looks looks exactly like what we fixed in that PR. It happens when you fly without airspeed and only switch into an altitude controlled mode once in air.
[image]

Please update to the latest 1.14 release https://github.com/PX4/PX4-Autopilot/releases/tag/untagged-e22115b9a03956561336

autoflightlab, 2024-03-18:
Thanks for taking a look at this.  Using Matlab UAV Toolbox, it appears that the pitch command steadily increases from the time the plane is armed rather than an abrupt jump when switching to RTL.  Using Flight Review, it shows the pitch command as zero and making a big jump when RTL engaged (first time) which would be more consistent with the earlier problem.

closed 2026-04-09 by github-actions[bot]