PX4/PX4-Autopilot #22796: [Bug] How to switch Crazyfly to Offboard mode in QGC

yuxuanwu1234567, 2024-02-23:
### Describe the bug

Unable to switch to off board mode in QGC. Do you need to make some configurations, such as topic forwarding in fixed-point mode to receive location information

### To Reproduce

Unable to switch

### Expected behavior

Switch off board mode and use motion capture data to control Crazyfly's indoor flight

### Flight Log

no

### Software Version

v1.12.0

### Flight controller

crazyflie2.1

### Vehicle type

None

yuxuanwu1234567, 2024-02-23:
@Jaeyoung-Lim

yuxuanwu1234567, 2024-02-23:
@davids5

Jaeyoung-Lim, 2024-02-23:
@yuxuanwu1234567 You cannot just switch to offboard mode. Please refer to the documentation on how to switch to offboard mode: https://docs.px4.io/main/en/flight_modes/offboard.html

yuxuanwu1234567, 2024-02-26:
> @yuxuanwu1234567 This is not a bug.
> 
> You cannot just switch to offboard mode. Please refer to the documentation on how to switch to offboard mode: https://docs.px4.io/main/en/flight_modes/offboard.html

Thank you for your reply. I first connected Crazyfly to the motion capture, then switched to position, and then switched from fixed point to off board, but it still hasn't been implemented.

Jaeyoung-Lim, 2024-02-26:
@yuxuanwu1234567 This is the expected behavior. You need to send offboard setpoints in order to ne able to switch to offboard mode

closed 2026-04-08 by github-actions[bot]