PX4/PX4-Autopilot #22818: [Bug] "differential_drive stop" produces Segmentation fault

slgrobotics, 2024-02-28:
### Describe the bug

When running a "_make px4_sitl gz_r1_rover_" (or _gz_lawnmower_) differential_drive module runs at startup by the rc.rover_differential_apps

If user wants to stop the module, by  "_differential_drive stop_" - a Segmentation fault happens and whole PX4 program quits.

@PerFrivik @MaEtUgR @dagar - issue created per your request

### To Reproduce

pxh> differential_drive stop

... Segmentation fault

### Expected behavior

The module should stop and could be started again

### Flight Log

no log needed

### Software Version

current main

### Flight controller

N/A - sitl

### Vehicle type

Rover

MaEtUgR, 2024-02-28:
Thanks for the issue! I started testing during the dev call and so far I found:
- The module stops fine just like expected when running and stopping it e.g. in an gz_x500 simulation where it's not in the loop. But stoping the `mc_rate_control` module which is AFAIK in the path for lockstep also works fine there.
- When not calling the exit and cleanup here: https://github.com/PX4/PX4-Autopilot/blob/28db3e1c8ca97fd1bb6067be15c9caf19db8e735/src/modules/differential_drive/DifferentialDrive.cpp#L68 it stops without segfault but blocks the shell for several seconds doing so. It can then be started and stopped again but I'm not sure if that's without residues.

PerFrivik, 2024-02-28:
Thanks, I'll take a look!

matteodelseppia, 2024-05-22:
I think I fixed the segmentation fault. You just have to add a return statement after the exit_and_cleanup() (line 68 of DifferentialDrive.cpp), otherwise it will try to update the current parameters even if the memory for the module has already been freed.
```
	if (should_exit()) {
		ScheduleClear();
		exit_and_cleanup();
                return; // < add this
	}
```

chfriedrich98, 2025-02-11:
Closing this issue since it references an outdated version of the differential rover module.
If this problem is present in the updated version, please open another issue.

closed 2025-02-11 by chfriedrich98