PX4/PX4-Autopilot #22819: [Bug] Rovers in SITL/GZ don't want to move when executing a mission

slgrobotics, 2024-02-28:
### Describe the bug

Rovers in SITL/GZ don't want to move when executing a mission, hesitate forever departing waypoints.

@PerFrivik @MaEtUgR @dagar - issue created per your request

### To Reproduce

Build a rover -  _"make px4_sitl gz_r1_rover"_ (or "_make px4_sitl gz_lawnmower_") and upload a mission.

Observe the rover to turn OK, but to hesitate (not move for long time) while departing any waypoint.

### Expected behavior

Robot should decisively execute any valid mission without long delays

### Screenshot / Media

Here is a possible mission (*.plan file for QGC), or you can create your own:

{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 1.3,
        "firmwareType": 12,
        "globalPlanAltitudeMode": 1,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    47.39822890941401,
                    8.545869140532716,
                    50
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    47.39823631964971,
                    8.546374332973812,
                    50
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 3,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    47.39781842273969,
                    8.546474618542533,
                    50
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 4,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    47.39781257113381,
                    8.545799130146975,
                    50
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 50,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 5,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    47.39799070583146,
                    8.545749396060955,
                    50
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            47.398055046493546,
            8.545940920918156,
            0
        ],
        "vehicleType": 10,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 2
    },
    "version": 1
}

### Flight Log

N/A

### Software Version

latest "main"

### Flight controller

N/A - sitl

### Vehicle type

Rover

PerFrivik, 2024-03-07:
Hey @slgrobotics sorry for not getting back to you earlier I have been traveling for business, but I will try to have a PR opened for this and #22818 by the next meeting on Wednesday.

chfriedrich98, 2024-09-13:
This bug has been fixed in https://github.com/PX4/PX4-Autopilot/pull/23430

closed 2024-09-13 by chfriedrich98