PX4/PX4-Autopilot #22821: Pixhawk 4 unable to complete serial connection with MicroXRCEAgent

CannonFodder511, 2024-02-28:
### Describe the bug

I am running a connection from a Pixhawk 4 running firmware 1.14.0. It is connected via a serial connection to a raspberry pi running Jammy and ROS 2 humble. This runs eProsima's MicroXRCEAgent. 

Things I've tried:

- There is nothing else trying to use the Telem 2 port on the Pixhawk.
- I am able to run mavproxy.py over the Telem 2 port
- A loopback script confirms the port is usable.

What else can I try to debug this?

Mavlink output:
`nsh> uxrce_dds_client start -t serial -d /dev/ttyS2 -b 115200
nsh> uxrce_dds_client status
INFO  [uxrce_dds_client] Running, disconnected
INFO  [uxrce_dds_client] Using transport: serial`

Pi Output:
`/ros_ws: MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v6
[1709157626.383849] info      |  TermiosAgentLinux.cpp  |   init                            |  running...                      |  fd: 3
[1709157626.385726] info      |  Root.cpp                   |   set_verbose_level          |  logger setup                |  verbose_level:  6`

### To Reproduce

Setup according to this page: https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html
Run "uxrce_dds_client start -t serial -d /dev/ttyS2 -b 115200" on the Pixhawk 4 over mavlink
Run  "MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v6" on the Pi

### Expected behavior

Pi and PX4 should connect, the PX4 posting topics to the XRCE Agent

### Flight Log

Mavlink output:
`nsh> uxrce_dds_client start -t serial -d /dev/ttyS2 -b 115200
nsh> uxrce_dds_client status
INFO  [uxrce_dds_client] Running, disconnected
INFO  [uxrce_dds_client] Using transport: serial`

Pi Output:
`/ros_ws: MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 115200 -v6
[1709157626.383849] info      |  TermiosAgentLinux.cpp  |   init                            |  running...                      |  fd: 3
[1709157626.385726] info      |  Root.cpp                   |   set_verbose_level          |  logger setup                |  verbose_level:  6`

### Software Version

1.14.0

### Flight controller

Pixhawk 4

### Vehicle type

None

mrpollo, 2024-02-28:
Make sure you there's nothing else running on the same port, it's either MAVLink or DDS the link can't handle both

CannonFodder511, 2024-02-28:
> Make sure you there's nothing else running on the same port, it's either MAVLink or DDS the link can't handle both

Is there any way to do this besides the parameters on QGC? All the mavlink parameters are set to disabled

chutsu, 2024-03-01:
Are you user you swapped the serial RX / TX?

CannonFodder511, 2024-03-02:
I followed the guide for the connections but did try swapping. Dial out permissions are enabled for the user and root

chutsu, 2024-03-02:
Also couple other things. 

- You haven't told us what RPi you're using
- I presume you followed this guide: https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html
- Why are your baud-rates different to the defaults?
- Have you tried simply communicating to TELEM2 as a MAVLink connection? (not using the USB port of the Pixhawk)

Geol-23, 2024-07-11:
Hello, 
I am having the same issue. I'm using a Raspberry Pi 4 (22.04) to communicate with an mRo Pixhawk (1.14.3). I've followed the [guide](https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html) and installed the agent as a standalone on the Pi, I've used the reccommended baud rates and ports (921600, TELEM2 on the FC and UART0 on the Pi) and I believe the pins are the correct way around. I've also added the dialout permissions and I'm pretty sure nothing else is running on that port.

To diagnose this I've tried to boot up another FC (same version as the main FC) and the issue persists. I've established communication using mavproxy according to the [guide](https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html).

I am getting the same logs per the issue. Any help is appreciated.
Cheers.

mrpollo, 2024-07-13:
@Geol-23 please open a new issue