PX4/PX4-Autopilot #22826: which python3 we want to use?

mengchaoheng, 2024-03-02:
### Describe the bug

when I use `MacOS Development Environment`,  after：
```
git clone https://github.com/PX4/PX4-Autopilot.git --recursive
cd PX4-Autopilot/Tools/setup
sh macos.sh
```

output is :

```Console
➜  PX4-Autopilot git:(9687324778) sh Tools/setup/macos.sh                
px4-dev already installed
Installing PX4 Python3 dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try brew install
    xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-brew-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-brew packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try brew install
    xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-brew-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip.
    
    If you wish to install a non-brew packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
All set! PX4 toolchain installed!
➜  PX4-Autopilot git:(9687324778) which python3
/usr/local/bin/python3
➜  PX4-Autopilot git:(9687324778) /usr/bin/python3 --version
Python 3.9.6
➜  PX4-Autopilot git:(9687324778) python3 --version         
Python 3.12.2
➜  PX4-Autopilot git:(9687324778) 
```

### To Reproduce

[following instructions set up a PX4 development environment for macOS](https://docs.px4.io/v1.13/en/dev_setup/dev_env_mac.html)

### Expected behavior

this instructions in  /Tools/setup/macos.sh will running:
```
python3 -m pip install future
python3 -m pip install --user -r ${DIR}/requirements.txt
```

### Flight Log

non

### Software Version

px4 v1.14.2, I want to build `px4_sitl` on macos sonoma 14.3.1 (m1). 

### Flight controller

non

### Vehicle type

None

mengchaoheng, 2024-03-02:
Python 3.12.2 is installed after `brew install px4-sim-gazebo`

mengchaoheng, 2024-03-02:
when I select python3.9, run all instructions in [MacOS Development Environment](https://docs.px4.io/v1.13/en/dev_setup/dev_env_mac.html), `make px4_sitl jmavsim` success. but `make px4_sitl gazebo-classic` failed:

```Console
➜  PX4-Autopilot git:(9687324778) ✗ make px4_sitl gazebo-classic                    

[0/10] git submodule Tools/simulation/gazebo-classic/sitl_gazebo-classic
[5/10] Performing configure step for 'sitl_gazebo-classic'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is AppleClang 15.0.0.15000100
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /usr/local/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.58") found components: system thread filesystem 
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.25
-- Found assimp: /usr/local/Cellar/assimp/5.3.1/include (found version "5.3.0") 
-- Found Threads: TRUE  
-- Found DART: /usr/local/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/local/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/lib/libz.tbd (found version "1.2.12")  
-- Looking for ignition-math6 -- found version 6.15.1
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib
-- Looking for ignition-math6 -- found version 6.15.1
-- Looking for ignition-transport8 -- found version 8.5.0
-- Searching for dependencies of ignition-transport8
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq, version 4.3.5
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
[... 137 lines cut ...]
[69/142] Linking CXX shared library libgazebo_groundtruth_plugin.dylib
ld: warning: ignoring duplicate libraries: '-lBulletCollision', '-lBulletDynamics', '-lBulletSoftBody', '-lLinearMath', '-ldl', '-lglib-2.0', '-lgobject-2.0', '-licudata', '-licuuc', '-lintl', '-lm'
[... cut ...]
FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build 
cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic && /usr/local/Cellar/cmake/3.28.3/bin/cmake --build /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -- -j 1
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
➜  PX4-Autopilot git:(9687324778) ✗ 
```

mengchaoheng, 2024-03-02:
another version:

```Console
➜  PX4-Autopilot git:(6823cbc414) ✗ make px4_sitl gazebo 
[0/10] git submodule Tools/sitl_gazebo
[5/10] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is AppleClang 15.0.0.15000100
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /usr/local/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.58") found components: system thread filesystem 
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.25
-- Found assimp: /usr/local/Cellar/assimp/5.3.1/include (found version "5.3.0") 
-- Found Threads: TRUE  
-- Found DART: /usr/local/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/local/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/lib/libz.tbd (found version "1.2.12")  
-- Looking for ignition-math6 -- found version 6.15.1
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib
-- Looking for ignition-math6 -- found version 6.15.1
-- Looking for ignition-transport8 -- found version 8.5.0
-- Searching for dependencies of ignition-transport8
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq, version 4.3.5
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.15.1
[... 260 lines cut ...]
^
/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_video_stream_widget.cpp:29:5: note: expanded from macro 'QT_MOC_LITERAL'
    Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET(len, \
    ^
/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_video_stream_widget.cpp:37:1: error: use of undeclared identifier 'Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET'
QT_MOC_LITERAL(2, 35, 0) // ""
^
[... cut ...]
[115/130] Building CXX object CMakeFiles/Fo...ual.dir/src/force_visual/force_visual.cpp.o
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.28.3/bin/cmake --build /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 2
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

mengchaoheng, 2024-03-02:
fmu-v5 error too:

```Console 
 PX4-Autopilot git:(6823cbc414) ✗ make px4_fmu-v5                        
[0/1143] git submodule src/drivers/uavcan/libuavcan
[1/1143] Running dsdl compiler
FAILED: libuavcan_dsdlc_run.stamp /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/libuavcan_dsdlc_run.stamp 
cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan && /usr/local/bin/python3 /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc test/dsdl_test/root_ns_a test/dsdl_test/root_ns_b /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/../dsdl/uavcan -Oinclude/dsdlc_generated && /usr/local/Cellar/cmake/3.28.3/bin/cmake -E touch /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/libuavcan_dsdlc_run.stamp
Traceback (most recent call last):
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/uavcan/__init__.py", line 15, in <module>
    import uavcan.transport as transport
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/uavcan/transport.py", line 203, in <module>
    class ArrayValue(BaseValue, collections.MutableSequence):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableSequence'
[2/1143] PX4 UAVCAN dsdl compiler
FAILED: src/drivers/uavcan/px4_uavcan_dsdlc_run.stamp /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/src/drivers/uavcan/px4_uavcan_dsdlc_run.stamp 
cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/src/drivers/uavcan && /usr/local/bin/python3 /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc --outdir /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/src/drivers/uavcan/include/dsdlc_generated /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/dsdl/ardupilot /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/dsdl/com /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/dsdl/cuav /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/dsdl/dronecan /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/dsdl/uavcan && /usr/local/Cellar/cmake/3.28.3/bin/cmake -E touch px4_uavcan_dsdlc_run.stamp
Traceback (most recent call last):
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 18, in <module>
    from uavcan import dsdl
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/uavcan/__init__.py", line 15, in <module>
    import uavcan.transport as transport
  File "/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/uavcan/transport.py", line 203, in <module>
    class ArrayValue(BaseValue, collections.MutableSequence):
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableSequence'
[5/1143] Generating /Users/mch/Proj/Mac_DF/...topilot/platforms/nuttx/NuttX/nuttx/.config
ninja: build stopped: subcommand failed.
make: *** [px4_fmu-v5] Error 1

```

mengchaoheng, 2024-03-08:
the error fix by 
```
brew uninstall --ignore-dependencies QT
```
But `make px4_sitl gazebo` still failed. 

```Console
➜  PX4-Autopilot git:(6823cbc414) ✗ make px4_sitl gazebo                                                            
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/tmp && /Users/mch/Proj/Mac_D... /Users/mch/Proj/Mac_DF/PX4-Autopilot /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot
build_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
gzserver(20729,0x204d8b240) malloc: Incorrect checksum for freed object 0x7f8f6cb475a0: probably modified after being freed.
Corrupt value: 0x4d555458
gzserver(20729,0x204d8b240) malloc: *** set a breakpoint in malloc_error_break to debug
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 217: 20729 Abort trap: 6           gzserver $verbose $world_path $ros_args
SITL COMMAND: "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mch/Proj/Mac_DF/PX4-Autopilot"/test_data
Creating symlink /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/mch/Proj/Mac_DF/PX4-Autopilo

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Err] [ConnectionManager.cc:121] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:78] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo
gcc^CERROR [px4] Startup script returned with return value: 2
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 260: kill: (20729) - No such process
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 262: kill: (20736) - No such process
ninja: build stopped: interrupted by user.
make: *** [px4_sitl] Interrupt: 2

➜  PX4-Autopilot git:(6823cbc414) ✗ gcc --version
Apple clang version 15.0.0 (clang-1500.1.0.2.5)
Target: x86_64-apple-darwin23.3.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
➜  PX4-Autopilot git:(6823cbc414) ✗ which python
python not found
➜  PX4-Autopilot git:(6823cbc414) ✗ which python3
/usr/local/bin/python3
➜  PX4-Autopilot git:(6823cbc414) ✗ python3 --version
Python 3.12.2
➜  PX4-Autopilot git:(6823cbc414) ✗ gazebo -v
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

➜  PX4-Autopilot git:(6823cbc414) ✗ 
```

mengchaoheng, 2024-03-08:
what is this?

```
gzserver(20729,0x204d8b240) malloc: Incorrect checksum for freed object 0x7f8f6cb475a0: probably modified after being freed.
Corrupt value: 0x4d555458
gzserver(20729,0x204d8b240) malloc: *** set a breakpoint in malloc_error_break to debug
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 217: 20729 Abort trap: 6           gzserver $verbose $world_path $ros_args

```

MBrandt-NASA, 2024-03-29:
> the error fix by
> 
> ```
> brew uninstall --ignore-dependencies QT
> ```
> 

I just want to thank you for sharing:

`brew uninstall --ignore-dependencies QT`

That totally fixed my PX4 gazebo-classic QT build errors on MacOS 13.6.1 Ventura... 

Now, 

`make px4_sitl gazebo-classic`

works like a champ.

anhtuanqn1002, 2024-04-05:
> what is this?
>
> ```
> gzserver(20729,0x204d8b240) malloc: Incorrect checksum for freed object 0x7f8f6cb475a0: probably modified after being freed.
> Corrupt value: 0x4d555458
> gzserver(20729,0x204d8b240) [...]

I have the same problem on MacOS Sonoma 14.4.1 M1.
[image]

Did you resolve it, @mengchaoheng ?

mengchaoheng, 2024-04-14:
@anhtuanqn1002 not yet, need help

mengchaoheng, 2024-04-14:
@MBrandt-NASA  any suggestion for our error ?

mengchaoheng, 2024-04-14:
if Comment out line 71 of `PX4-Autopilot/cmake/px4_add_common_flags.cmake`:
```
# -Werror
```

`make px4_fmu-v5` will work:
```
➜  PX4-Autopilot git:(heads/v1.13.3) ✗ make px4_fmu-v5
[0/1162] git submodule src/modules/mavlink/mavlink
[8/1162] git submodule src/drivers/gps/devices
[13/1162] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /Users/mch/Proj/Mac_DF/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 229 MAVLink message types in 4 XML files
Generating C implementation in directory /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/development
Generating C implementation in directory /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/standard
Generating C implementation in directory /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/common
Generating C implementation in directory /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/mavlink/minimal
Copying fixed headers for protocol 2.0 to /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/mavlink
[15/1162] git submodule src/drivers/uavcan/libuavcan
[16/1162] git submodule platforms/nuttx/NuttX/nuttx
[19/1162] Running dsdl compiler
/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/pyratemp.py:273: SyntaxWarning: invalid escape sequence '\#'
  """Replace special characters by their escape sequence.
[83/1162] Generating component_general.json and checksums.h
/Users/mch/Proj/Mac_DF/PX4-Autopilot/src/lib/component_information/generate_component_general.py:23: SyntaxWarning: invalid escape sequence '\s'
  version_search = re.search('PX4_GIT_TAG_OR_BRANCH_NAME\s+"(.+)"', line)
[1160/1162] Linking CXX executable px4_fmu-v5_default.elf
Memory region         Used Size  Region Size  %age Used
      FLASH_ITCM:          0 GB      2016 KB      0.00%
      FLASH_AXIM:     1976721 B      2016 KB     95.75%
        ITCM_RAM:          0 GB        16 KB      0.00%
        DTCM_RAM:          0 GB       128 KB      0.00%
           SRAM1:       45544 B       368 KB     12.09%
           SRAM2:          0 GB        16 KB      0.00%
[1162/1162] Creating /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_fmu-v5_default/px4_fmu-v5_default.px4
➜  PX4-Autopilot git:(heads/v1.13.3) ✗  ```

mengchaoheng, 2024-04-14:
for python I use `--break-system-packages `:
```
$  python3 -m pip install --user xxx --break-system-packages 
```
to run ` python3 -m pip install --user xxx`. and for QT:
```
 brew list --versions qt     
brew list --versions pyqt
```
and ubinstall all qt and pyqt except @5, for me that is :
```
brew uninstall qt
 brew uninstall pyqt 
```
and then:
```
 brew install  pyqt@5
brew install qt@5 
```

mengchaoheng, 2024-04-14:
now still have error:
```
 PX4-Autopilot git:(heads/v1.13.3) ✗ make px4_sitl gazebo           
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/tmp && /Users/mch/Proj/Ma...sers/mch/Proj/Mac_DF/PX4-Autopilot /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot
build_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
gzserver(24755,0x204aa5240) malloc: Incorrect checksum for freed object 0x7fc70c1af470: probably modified after being freed.
Corrupt value: 0x4d555458
gzserver(24755,0x204aa5240) malloc: *** set a breakpoint in malloc_error_break to debug
/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_run.sh: line 217: 24755 Abort trap: 6           gzserver $verbose $world_path $ros_args
SITL COMMAND: "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mch/Proj/Mac_DF/PX4-Autopilot"/test_data
Creating symlink /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/mch/Proj/Mac_DF/PX4-Autopilo

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560

```

mengchaoheng, 2024-04-14:
and headless:
```
➜  PX4-Autopilot git:(heads/v1.13.3) ✗ HEADLESS=1 make px4_sitl gazebo

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/tmp && /Users/mch/Proj/Ma...sers/mch/Proj/Mac_DF/PX4-Autopilot /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot
build_path: /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /Users/mch/Proj/Mac_DF/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
not running gazebo gui
SITL COMMAND: "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/mch/Proj/Mac_DF/PX4-Autopilot"/test_data
Creating symlink /Users/mch/Proj/Mac_DF/PX4-Autopilot/build/px4_sitl_default/etc -> /Users/mch/Proj/Mac_DF/PX4-Autopilo

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
gzserver(24906,0x204460240) malloc: Incorrect checksum for freed object 0x7f8a30bc9ae8: probably modified after being freed.
Corrupt value: 0x20a800000000
gzserver(24906,0x204460240) malloc: *** set a breakpoint in malloc_error_break to debug

```

mengchaoheng, 2024-04-20:
I solved this problem, on macos 14.3.1. everything is default, 

summarized below

1. Regarding python3, macos comes with python3, but when installing px4-sim-gazebo, extra python3 (3.11 and 3.12) will also be installed. At this time, either use the python virtual environment or delete the extra python3.
```
brew uninstall --ignore-dependencies python3
brew uninstall --ignore-dependencies python@3.11 # option
```
But in fact you can use python3.12 , you must ensure that the python3 used in the following instructions is the same as when compiling `make px4_sitl gazebo`:
1).
```
python3 -m pip install --user pyserial empty toml numpy pandas jinja2 pyyaml pyros-genmsg packaging kconfiglib future jsonschema
```
2).
```
sh macos.sh
```
When you buid the code, you will see :
```
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.9.6", minimum required is "3")

```
You need to ensure that the above python version is consistent!!!

2. Regarding QT, we only need qt5 and pyqt5. When installing px4-sim-gazebo, a higher version (6) will be installed by default.
```
brew uninstall --ignore-dependencies qt
brew uninstall --ignore-dependencies pyqt
brew install  pyqt@5  # if don't have
brew install qt@5  # if don't have
```
if they install new version python, just run the follow again:
```
brew uninstall --ignore-dependencies python3
```
3. Regarding SDK, use complete xcode instead of CLT.
you can check you path:
```
xcode-select -p
```
you maybe see the follow:
```
/Library/Developer/CommandLineTools

```

if that happen, install xcode by app store and change the path:
```
sudo xcode-select --switch /Applications/Xcode.app
```

mengchaoheng, 2024-04-27:
some ref. https://github.com/PX4/PX4-SITL_gazebo-classic/issues/1021#issuecomment-2078922511

MeethiKishori, 2025-05-18:
> I solved this problem, on macos 14.3.1. everything is default,
> 
> summarized below
> 
> 1. Regarding python3, macos comes with python3, but when installing px4-sim-gazebo, extra python3 (3.11 and 3.12) will also be installed. At this time, either use the python virtual environment or delete the extra python3.
> 
> ```
> brew uninstall --ignore-dependencies python3
> brew uninstall --ignore-dependencies python@3.11 # option
> ```
> 
> But in fact you can use python3.12 , you must ensure that the python3 used in the following instructions 1) and 2) is the same as when compiling `make px4_sitl gazebo`: 1).
> 
> ```
> python3 -m pip install --user pyserial empty toml numpy pandas jinja2 pyyaml pyros-genmsg packaging kconfiglib future jsonschema
> ```
> 
> 2).
> 
> ```
> sh macos.sh
> ```
> 
> When you buid the code, you will see :
> 
> ```
> -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.9.6", minimum required is "3")
> ```
> 
> You need to ensure that the above python version is consistent!!! for ensure this, you can install `brew install px4-sim-gazebo ` before Install the required Python packages, and then you will use python3.12
> 
> 2. Regarding QT, we only need qt5 and pyqt5. When installing px4-sim-gazebo, a higher version (6) will be installed by default.
> 
> ```
> brew uninstall --ignore-dependencies qt
> brew uninstall --ignore-dependencies pyqt
> brew install  pyqt@5  # if don't have
> brew install qt@5  # if don't have
> ```
> 
> if they install new version python, just run the follow again:
> 
> ```
> brew uninstall --ignore-dependencies python3
> ```
> 
> 3. Regarding SDK, use complete xcode instead of CLT.
>    you can check you path:
> 
> ```
> xcode-select -p
> ```
> 
> you maybe see the follow:
> 
> ```
> /Library/Developer/CommandLineTools
> ```
> 
> if that happen, install xcode by app store and change the path:
> 
> ```
> sudo xcode-select --switch /Applications/Xcode.app
> ```

@mengchaoheng thanks for your research on the issue, i was lost in the same issue for a week now until i saw this issue. 
Can i use default python3.11 and perform all installs? How to use a Venv to ease the task ?

mengchaoheng, 2025-10-17:
@mrpollo Thank you for your reply, I think the problem mainly lies in px4-sim.

closed 2026-04-08 by github-actions[bot]