PX4/PX4-Autopilot #22832: QGC is not detecting PX4Flow sensor connected through pixhawk2.4.8 with PX4 software

NithinKrishnan28, 2024-03-04:
### Describe the bug

I am trying to connect my px4 flow sensor with px4. I updated the firmware on px4 flow sensor by directly connecting it to the QGC using USB. But when i connect to the Pixhawk and then try to configure it, its not showing the px4 flow sensor connected. Also, when i connect px4 flow directly it shows the error message that is shown in the below imaage.

[image]

Can anyone please suggest me what could be done to solve the above mentioned issue.
Thanking you in advance!!!!!

### To Reproduce

Drone not yet connected

### Expected behavior

The sensor is not detected.

### Screenshot / Media

[image]

### Flight Log

doing initial setup

### Software Version

latest firmware

### Flight controller

pixhawk 2.4.8

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Using I2C port to connect pixhawk and px4 flow sensor.

fernando-granco, 2024-03-09:
I was reading the documentation and remembered this. it might have something to do with it:

From: https://docs.px4.io/main/en/sensor/px4flow.html

PX4 does not support the PX4Flow sensor from PX4 v1.13.

Note:

PX4 v1.13 does not work with current PX4FLow firmware. PX4 may work with older PX4Flow firmware.

AsfandYar-Ktk, 2025-12-01:
Hi. I have the same issue. It would be a lot of help if you were able to find a way to make support for px4flow for the current px4 version. 
If not what was your solution to this problem?