PX4/PX4-Autopilot #22839: [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise

henrykotze, 2024-03-05:
### Describe the bug

A simple takeoff, hover (remaining in its position) and then landing successfully, contains 'derivative' type spike noise in the groundtruth readings when simulating a x500 in gazebo (gz). After a landing then preflight errors occur such as Yaw Estimate Error and GPS vertical drift. 

I am simulating this on a source built version of gz-sim7 on today's `main`: https://github.com/PX4/PX4-Autopilot/commits/9cef8346245b07c81dd52e679e0283eefaf66d97

Im working on a other branch which is based on a few months ago `main`, which has similiar 'derivative type spike noise in the groundtruth readings, but these was much less and though not ideal, it was "ignorable". 

### To Reproduce

Takeoff, Hover: (Should see some instability during hover in the attitude), Land. -> 

### Expected behavior

Little to no instability in the attitude of the drone.

### Screenshot / Media

Today's main
[image]

Similiar flight from a few months old branch, based on [a6d2c2cf5e21ecdad8d737c31de81fab128898c5](https://github.com/PX4/PX4-Autopilot/commit/a6d2c2cf5e21ecdad8d737c31de81fab128898c5)
[image]

### Flight Log

Flight Review Dowb>

### Software Version

[9cef8346245b07c81dd52e679e0283eefaf66d97](https://github.com/PX4/PX4-Autopilot/commits/9cef8346245b07c81dd52e679e0283eefaf66d97)

### Flight controller

SITL

### Vehicle type

Multicopter

frede791, 2024-03-07:
There was recently an update to the x500 where I added simulated IMU noise. https://github.com/PX4/PX4-gazebo-models/pull/34
Can you check whether you still get the same problem when you revert to the previous version of the x500?

slgrobotics, 2024-03-11:
I am having similar spikes when logging GPS velocities in SITL. I  think this started after recent changes in the world SDF files.

[image]

The plot is from a custom DebugArray - I subscribe to _sensor_gps_ message and just log the "vel_*" members. 

EKF speed (red line) ~1.5 m/s is correct, spikes to 30 are unrealistic. This is a slow rover.

frede791, 2024-03-13:
Hmm the noise model has changed slightly from what it was before. I took care to take the same parameter values wherever possible. @Jaeyoung-Lim any idea?

henrykotze, 2024-03-28:
Tested on main today, and same issue. 

@frede791 I also tested multiple commits before the simulated IMU noise, and same issue as describe.

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T23:03:26
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise (Issue #22839)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T23:10:15
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise (Issue #22839)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

henrykotze, 2024-03-28:
No not yet.

Just for clarity. 
I have tested the following commits for the gz submodule and all contain the describe bug:
[6b4ed09d1b495fbff663f098979cc046df013abd](https://github.com/PX4/PX4-gazebo-models/commits/6b4ed09d1b495fbff663f098979cc046df013abd/)

[image]

[953e02b5362f9e7d0fd1a6a85e0ff60a32c99942](https://github.com/PX4/PX4-gazebo-models/commits/953e02b5362f9e7d0fd1a6a85e0ff60a32c99942/)
[image]

[222833656802532ec2271986a65fd198cfa48259](https://github.com/PX4/PX4-gazebo-models/commits/222833656802532ec2271986a65fd198cfa48259/)
[image]

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T23:28:17
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise (Issue #22839)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

Jaeyoung-Lim, 2024-03-28:
@henrykotze It would be easier if we can pinpoint which change introduced the problem :smile:

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T23:31:42
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise (Issue #22839)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

henrykotze, 2024-03-28:
@Jaeyoung-Lim  Agreed, :+1: , just added those for clarity.

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T23:33:44
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise (Issue #22839)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

henrykotze, 2024-03-28:
According to git bisect its [bb53781b8f94fdc3b3d94c23dda429f0932c28fc](https://github.com/PX4/PX4-Autopilot/commit/bb53781b8f94fdc3b3d94c23dda429f0932c28fc) , which I have confirmed by testing the previous commit which does not contain the bug.

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-29T00:37:56
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Groundtruth readings from Gazebo Contains Spiky Derivative Type Noise (Issue #22839)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

dagar, 2024-04-02:
We're computing the angular velocity ground truth ourselves here. https://github.com/PX4/PX4-Autopilot/blob/8579175013a991fb295ebb19f2258f2f1ccdbb49/src/modules/simulation/gz_bridge/GZBridge.cpp#L570-L575

Can you verify if `vehicle_attitude_groundtruth` looks clean? Otherwise check the `dt`. https://github.com/PX4/PX4-Autopilot/blob/8579175013a991fb295ebb19f2258f2f1ccdbb49/src/modules/simulation/gz_bridge/GZBridge.cpp#L532-L533

henrykotze, 2024-04-03:
@dagar vehicle_attitude_groundtruth looks clean, but `dt` varied alot.
This led to the following fix:
Fix implemented: https://github.com/PX4/PX4-Autopilot/pull/22967

closed 2024-04-04 by dagar