PX4/PX4-Autopilot #22857: [Bug] PIXHAWK6X-RT's SD write process can exceed 80 ms.

muramura, 2024-03-09:
### Describe the bug

I did SD_BENCH with microSD SANDISK 32GB Extreme PRO on PIXHAWK6X-RT.
I have used this MICRO SD with good results with PIXHAWK5X and 6X.
A few times out of 50 times, the write time exceeds 100 milliseconds.

### To Reproduce

1. insert microSD SANDISK 32GB Extreme PRO into PIXHAWK6X-RT.
2. Connect QGC to PIXHAWK6X-RT.
3. Execute "sd_bench -r 50" from MAVLINK CONSOLE.

### Expected behavior

The write time must be within 20 ms.

### Screenshot / Media

[image]

### Flight Log

none

### Software Version

nsh> ver all
HW arch: PX4_FMU_V6XRT
HW type: V6XRT
HW FMUM ID: 0x000
HW BASE ID: 0x000
PX4 git-hash: ff3c63fd2d89f703093e85ce261d2d3918a7f379
PX4 version: 1.15.0 0 (17760256)
PX4 git-branch: PX4_0207-4
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 693b9e782535f12e6a4ab657c7a0c3bd92b45fb1
Build datetime: Mar  9 2024 09:28:41
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000900000000000000008292b37f162b580e
MCU: i.MX RT1170 rB0, rev. 0

### Flight controller

PIXHAWK6X-RT

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

simple substance

### Additional context

none

muramura, 2024-03-09:
SD_BENCH was performed with a "Kioxia" MICRO SD card inserted in PIXHAWK6X-RT.

[image]

PetervdPerk-NXP, 2024-03-11:
SD card performance and max write time is mostly card dependent.
This is because the SD card itself does the wear-leveling at intervals we can't control.
Also the state of the flash on the sd card itself determines this.

I think the more important question is does the PX4 logging work as expected?

muramura, 2024-03-16:
SANDISK EXTREME MICROSDHC UHS-I CARD R100 W60MB/S

nsh> sd_bench -r 100
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   155.94 KB/s, max write time: 46 ms (=  86.96 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  1:   160.55 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  2:   158.14 KB/s, max write time: 57 ms (=  70.18 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  3:   160.39 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  4:   160.61 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run  5:   157.49 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  6:   160.91 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  7:   157.65 KB/s, max write time: 60 ms (=  66.67 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run  8:   161.18 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  9:   161.10 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 10:   158.10 KB/s, max write time: 45 ms (=  88.89 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 11:   161.12 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 12:   155.56 KB/s, max write time: 58 ms (=  68.97 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 13:   161.38 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 14:   161.40 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 15:   157.91 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 16:   160.98 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 17:   158.60 KB/s, max write time: 48 ms (=  83.33 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 18:   159.22 KB/s, max write time: 40 ms (= 100.00 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 19:   158.91 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 46 ms
INFO  [sd_bench]   Run 20:   160.38 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 21:   161.49 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 22:   157.73 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 23:   160.59 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 24:   158.02 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 25:   158.82 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 26:   161.32 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 27:   158.56 KB/s, max write time: 57 ms (=  70.18 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 28:   160.35 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 29:   158.50 KB/s, max write time: 67 ms (=  59.70 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 30:   160.27 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 31:   161.24 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 32:   158.86 KB/s, max write time: 59 ms (=  67.80 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 33:   160.94 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 34:   158.69 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 35:   161.26 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 36:   160.78 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 37:   157.79 KB/s, max write time: 57 ms (=  70.18 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 38:   158.63 KB/s, max write time: 36 ms (= 111.11 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 39:   158.30 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 40:   161.07 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 41:   157.54 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 42:   160.93 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 43:   160.82 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 44:   158.62 KB/s, max write time: 59 ms (=  67.80 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 45:   160.55 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 46:   158.38 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 47:   160.32 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 48:   161.11 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 49:   157.59 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 50:   159.81 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 51:   158.18 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 52:   161.16 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 53:   161.66 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 54:   158.72 KB/s, max write time: 58 ms (=  68.97 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 55:   158.93 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 20 ms
[... cut ...]
INFO  [sd_bench]   Run 95:   157.74 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 96:   161.72 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 97:   160.38 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 98:   158.11 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 99:   160.46 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 15 ms
INFO  [sd_bench]   Avg   :   159.58 KB/s
INFO  [sd_bench]   maxMWT:       67 ms
nsh>

muramura, 2024-03-16:
BUFFALO MICROSDHC UHS-I U1/V10 CLASS 10/AI 32GB

nsh> sd_bench -r 100
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   232.66 KB/s, max write time: 90 ms (=  44.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  1:   385.83 KB/s, max write time: 71 ms (=  56.34 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:   212.72 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  3:   383.82 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  4:   226.73 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  5:   347.10 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  6:   271.01 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  7:   302.92 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  8:   309.02 KB/s, max write time: 71 ms (=  56.34 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  9:   267.43 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 10:   349.99 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 11:   220.83 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:   397.80 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 13:   211.43 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 14:   371.33 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 15:   238.65 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 16:   337.62 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 17:   279.12 KB/s, max write time: 71 ms (=  56.34 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 18:   293.13 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 19:   320.79 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 20:   248.14 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 62 ms
INFO  [sd_bench]   Run 21:   373.10 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 22:   211.15 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 23:   404.89 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 24:   212.89 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 25:   353.09 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 26:   260.95 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 27:   313.30 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 28:   303.17 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 29:   270.46 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 30:   341.70 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 31:   232.72 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 32:   386.53 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 33:   210.57 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 34:   382.45 KB/s, max write time: 122 ms (=  32.79 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 35:   226.30 KB/s, max write time: 71 ms (=  56.34 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 36:   342.03 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 37:   271.19 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 38:   306.18 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 39:   318.58 KB/s, max write time: 71 ms (=  56.34 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 40:   259.32 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 41:   355.86 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 42:   216.97 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 43:   410.28 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 44:   210.73 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 45:   363.29 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 46:   256.58 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 47:   314.35 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 48:   306.60 KB/s, max write time: 73 ms (=  54.79 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 49:   271.00 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 50:   350.70 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 51:   224.00 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 52:   395.66 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 53:   214.37 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 54:   372.26 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 55:   248.38 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 56:   328.54 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 3 ms
[... cut ...]
INFO  [sd_bench]   Run 95:   204.73 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 96:   358.22 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 97:   256.71 KB/s, max write time: 71 ms (=  56.34 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 98:   312.55 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 99:   303.11 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 4 ms
INFO  [sd_bench]   Avg   :   298.36 KB/s
INFO  [sd_bench]   maxMWT:      122 ms
nsh>

muramura, 2024-03-16:
KIOXIA EXCERIA MICROSDHC UHS-I 32GB

nsh> sd_bench -r 100
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   780.20 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  1:   778.07 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run  2:   777.79 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run  3:   778.75 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run  4:   779.68 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run  5:   777.66 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run  6:   779.97 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run  7:   774.44 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  8:   779.44 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  9:   778.42 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 10:   777.72 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 11:   782.36 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 12:   780.33 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 13:   632.07 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 14:   781.99 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 15:   779.86 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 16:   776.51 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 17:   781.73 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 18:   778.43 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 19:   775.15 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 20:   782.24 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 21:   779.89 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 22:   775.30 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 23:   778.45 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 24:   782.80 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 25:   778.25 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 26:   777.53 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 27:   783.64 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 28:   775.68 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 29:   777.40 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 30:   779.87 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 31:   636.39 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 32:   779.15 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 33:   781.57 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 34:   771.44 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 35:   778.82 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 36:   775.83 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 37:   773.08 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 38:   781.02 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 39:   777.54 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 40:   779.35 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 41:   783.44 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 42:   777.95 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 43:   780.54 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 44:   777.58 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 45:   775.18 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 46:   780.02 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 47:   779.15 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 48:   656.58 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 127 ms
INFO  [sd_bench]   Run 49:   776.50 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 50:   778.00 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 51:   777.74 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 52:   781.38 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 53:   778.93 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 54:   777.49 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 55:   779.82 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 56:   779.55 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
[... cut ...]
INFO  [sd_bench]   Run 95:   779.73 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 96:   776.80 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 97:   778.03 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 98:   779.87 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 2 ms
INFO  [sd_bench]   Run 99:   777.61 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 2 ms
INFO  [sd_bench]   Avg   :   771.33 KB/s
INFO  [sd_bench]   maxMWT:      118 ms
nsh>

muramura, 2024-03-16:
SANDISK EXTREME PRO MICROSDHC UHS-I 32GB
Not Refomat

nsh> sd_bench -r 100
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   583.71 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  1:   578.67 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:   583.84 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  3:   584.54 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  4:   578.28 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  5:   584.67 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  6:   582.74 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  7:   583.51 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  8:   571.75 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  9:   581.70 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 10:   584.06 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 11:   580.34 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:   582.54 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 13:   582.78 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 14:   583.07 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 15:   581.22 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 16:   584.52 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 17:   584.09 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 18:   580.81 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 19:   583.23 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 20:   581.51 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 21:   583.17 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 22:   577.92 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 23:   581.41 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 24:   580.74 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 25:   579.34 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 26:   583.82 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 27:   582.15 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 28:   584.66 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 29:   578.65 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 30:   582.53 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 31:   579.65 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 32:   579.96 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 33:   580.44 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 34:   583.99 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 35:   581.71 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 36:   572.37 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 37:   582.70 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 38:   579.88 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 39:   580.21 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 40:   577.24 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 41:   585.10 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 42:   583.60 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 43:   583.03 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 44:   581.56 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 45:   583.88 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 46:   581.58 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 47:   580.43 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 48:   582.95 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 49:   579.93 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 50:   581.88 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 51:   580.96 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 52:   581.81 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 53:   577.25 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 54:   584.20 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 55:   581.87 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 56:   581.98 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
[... cut ...]
INFO  [sd_bench]   Run 95:   581.56 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 96:   580.05 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 97:   583.79 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 98:   582.63 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 99:   580.83 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Avg   :   581.56 KB/s
INFO  [sd_bench]   maxMWT:       32 ms
nsh>

muramura, 2024-03-17:
I have determined that the following MICROSD card is best suited for PIXHAWK6X-RT
I close.

SANDISK EXTREME PRO MICROSDHC UHS-I 32GB

closed 2024-03-17 by muramura