PX4/PX4-Autopilot #22858: [Bug] 64M bytes of flash memory not available in PIXHAWK6X-RT

muramura, 2024-03-09:
### Describe the bug

PIXHAWK6X-RT implements 64 Mbytes of flash memory.
However, a verification error occurs when writing to it.

### To Reproduce

1. Set the flash memory size to 64M bytes.
2. Build.
3. Upload to PIXHAWK6X-RT

### Expected behavior

Completing a successful verification

### Screenshot / Media

[image]

### Flight Log

nano

### Software Version

none

### Flight controller

PIXHAWK6X-RT

### Vehicle type

Multicopter

muramura, 2024-03-09:
If the flash memory size is 28M bytes, verification will be completed successfully.

[image]

nsh: sysinit: fopen failed: No such file or directory

NuttShell (NSH) NuttX-11.0.0
nsh> ver  all
HW arch: PX4_FMU_V6XRT
HW type: V6XRT
HW FMUM ID: 0x000
HW BASE ID: 0x000
PX4 git-hash: ff3c63fd2d89f703093e85ce261d2d3918a7f379
PX4 version: 1.15.0 0 (17760256)
PX4 git-branch: PX4_0207-4
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 693b9e782535f12e6a4ab657c7a0c3bd92b45fb1
Build datetime: Mar  9 2024 09:28:41
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000900000000000000008292b37f162b580e
MCU: i.MX RT1170 rB0, rev. 0
nsh>

PetervdPerk-NXP, 2024-03-11:
Here's a branch that extend the linker and bl to 64M.
https://github.com/PetervdPerk-NXP/Firmware/tree/pr-v6xrt-bootloader-64m
Again as discussed on the discord, we don't see an immediate reason to do this as the default.
Due to the increased flashing/update time and that current builds don't get even close to the 4M that's imposed right now.

muramura, 2024-03-11:
@PetervdPerk-NXP san.  Timeout occurs in verify.

[1207/1210] Linking CXX executable px4_fmu-v6xrt_default.elf
Memory region         Used Size  Region Size  %age Used
           flash:     2163752 B     65408 KB      3.23%
            sram:       92080 B      1280 KB      7.03%
            itcm:      245280 B       256 KB     93.57%
            dtcm:         936 B       256 KB      0.36%
[1209/1210] uploading px4
Waiting for bootloader...

Found board id: 35,0 bootloader version: 5 on /dev/serial/by-id/usb-Dronecode_Project__Inc._PX4_BL_FMU_v6XRT.x_0-if00
Loaded firmware for board id: 35,0 size: 2163752 bytes (3.23%) 

sn: 8292b37f162b580e31313630
chip: 001170b0
family: b'i.MX RT1170'
revision: b'rB0'
flash: 66977792 bytes
Windowed mode: False

Erase  : [====================] 100.0% (timeout: 14 seconds) 
Program: [====================] 100.0%
Verify : [                    ] 1.0%
ERROR: timeout waiting for data (4 bytes)
FAILED: platforms/nuttx/CMakeFiles/upload /home/muramura/work2/PX4-Autopilot/build/px4_fmu-v6xrt_default/platforms/nuttx/CMakeFiles/upload 
cd /home/muramura/work2/PX4-Autopilot/build/px4_fmu-v6xrt_default && /usr/bin/python3 /home/muramura/work2/PX4-Autopilot/Tools/px_uploader.py --port "/dev/serial/by-id/*-Dronecode_Project,_Inc._PX4_FMU_v6XRT.x*,/dev/serial/by-id/*PX4_BL*,/dev/serial/by-id/*BL_FMU*,/dev/serial/by-id/usb-The_Autopilot*,/dev/serial/by-id/usb-Bitcraze*,/dev/serial/by-id/pci-Bitcraze*,/dev/serial/by-id/usb-Gumstix*,/dev/serial/by-id/usb-Hex_ProfiCNC*,/dev/serial/by-id/usb-UVify*,/dev/serial/by-id/usb-ArduPilot*" /home/muramura/work2/PX4-Autopilot/build/px4_fmu-v6xrt_default/px4_fmu-v6xrt_default.px4
ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_fmu-v6xrt_default] Error 1
muramura@TUF:~/work2/PX4-Autopilot$

muramura, 2024-03-11:
Thank you very much.
I am happy.
I was able to upload the file successfully.
Furthermore, it worked fine.

[image]

[image]

PetervdPerk-NXP, 2024-03-11:
I got px_loader.py to work (but not reliable) by increasing sleeps see the [HACK] commit in 
https://github.com/PetervdPerk-NXP/Firmware/tree/pr-v6xrt-bootloader-64m

I think it has to do that with 64M enabled that verify takes way longer.
@davids5 any ideas?

davids5, 2024-03-11:
@PetervdPerk-NXP it is just a dumb wait so I am sure it will break. Possible scale it by returned size of flash

muramura, 2024-03-16:
Thank you. I have resolved the issue and will close it.

closed 2024-03-16 by muramura