PX4/PX4-Autopilot #22901: [Bug] CUAV 6X board detection sometimes fails on v1.14.x

julianoes, 2024-03-18:
### Describe the bug

Sometimes, the board version/revision check for the CUAV 6X fails. I observe this about 1 in 5 reboots, only with v1.14.x.

When it happens, the baros and mag are not getting initialized.

FYI @davids5, @CUAVcaijie 

### To Reproduce

1. Flash v1.14.x from QGC.
2. Check for baros and mag, or check `ver all`

You can also add a break point in:
https://github.com/PX4/PX4-Autopilot/blob/32aa3263a60d48a960eb8a2ccc50073815250889/platforms/nuttx/src/px4/stm/stm32_common/board_hw_info/board_hw_rev_ver.c#L335

And check `dn`. It changes slightly on every reboot.

### Expected behavior

Correct:

```
ver all
HW arch: PX4_FMU_V6X
HW type: V6X002001
HW version: 0x002
HW revision: 0x001
PX4 git-hash: b8c541dd7277ed735139d7d1bfb829d61fbe29fb
PX4 version: Release 1.14.0 (17694975)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: Oct 19 2023 16:24:32
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000313732373330511200400036
MCU: STM32H7[4|5]xxx, rev. V

```

Failed:
```
ver all
HW arch: PX4_FMU_V6X
HW type: V6X002000
HW version: 0x002
HW revision: 0x000
PX4 git-hash: b8c541dd7277ed735139d7d1bfb829d61fbe29fb
PX4 version: Release 1.14.0 (17694975)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: Oct 19 2023 16:24:32
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000313732373330511200400036
MCU: STM32H7[4|5]xxx, rev. V
```

### Flight Log

No flying required

### Software Version

v1.14.0

### Flight controller

CUAV 6X

### Vehicle type

None

davids5, 2024-03-18:
@julianoes The dn will fluctuate because of noise.  Would you capture the values on your HW when it is good an bad and post them here.

julianoes, 2024-03-18:
I remember seeing values between 15 and 21 but I don't have it set up anymore.

davids5, 2024-03-18:
> I remember seeing values between 15 and 21 but I don't have it set up anymore.

hmm both those values fall in the [1 bucket.](https://github.com/PX4/PX4-Autopilot/blob/main/platforms/nuttx/src/px4/stm/stm32_common/board_hw_info/board_hw_rev_ver.c#L94)

julianoes, 2024-03-18:
```
dn: 15
...
HW arch: PX4_FMU_V6X
HW type: V6X
HW FMUM ID: 0x000
HW BASE ID: 0x002
```

```
dn: 19
...
HW arch: PX4_FMU_V6X
HW type: V6X
HW FMUM ID: 0x001
HW BASE ID: 0x002
```

davids5, 2024-03-18:
Can you print it here https://github.com/PX4/PX4-Autopilot/blob/32aa3263a60d48a960eb8a2ccc50073815250889/platforms/nuttx/src/px4/stm/stm32_common/board_hw_info/board_hw_rev_ver.c#L87

davids5, 2024-03-18:
Ok I can see that dn scaling is seeing 15 as a 0 @CUAVcaijie What tolerance R's did you use?