PX4/PX4-Autopilot #22926: [Bug] Fixed-wing unable to disarm without GPS

MaEtUgR, 2024-03-25:
### Describe the bug

I went maiden flying my newly built Multiplex FunCub NG without GPS unit yet but with PX4 to handle Crossfire input, 7 PWM mixing (throttle, elevator, rudder, 2 ailerons, 2 flaps), gyro and attitude stabilization. GPS I'll build in later.
I armed, flew for a bit, landed, flipped the disarm switch and assumed the propeller will not turn but sometimes it unexpectedly started spinning again when the throttle stick was not at zero and in the end I had to disconnect the battery while it was still armed.

Note: It didn't always deny disarming. After some short flights it would work as expected.

### To Reproduce

It's also reproducible in simulation, see steps and log below.
1. Fixed-wing with no GPS
   Sim: `make px4_sitl gazebo-classic_plane` and execute `failure gps off` to disable GPS. 
2. Takeoff and fly in Stabilized, Acro, Manual
3. Land again
4. Try to disarm

### Expected behavior

For safety reasons I expect a guaranteed way to disarm the vehicle after landing.

### Flight Log

From the logs I found out what the chain of problems is:
- For fixed-wing unlike multicopter in Stabilized, Manual or Acro mode it's not allowed to disarm when the land detector doesn't declare the vehicle to be landed:
https://github.com/PX4/PX4-Autopilot/blob/bcbae86b9f869414866238a939232c5771e0f695/src/modules/commander/Commander.cpp#L641
- Land detector is not landed in the logs and after some inspection I'm pretty sure it's because it sees a horizontal velocity that is constantly above a certain threshold:
https://github.com/PX4/PX4-Autopilot/blob/bcbae86b9f869414866238a939232c5771e0f695/src/modules/land_detector/FixedwingLandDetector.cpp#L114-L115
- The estimator presumably publishes a best guess dead-reckon velocity but explicitly flags it as invalid using https://github.com/PX4/PX4-Autopilot/blob/bcbae86b9f869414866238a939232c5771e0f695/msg/VehicleLocalPosition.msg#L9 The land detector ignores that flag and uses the values in `vehicle_local_position.vx/vy` if they are not NAN:
https://github.com/PX4/PX4-Autopilot/blob/bcbae86b9f869414866238a939232c5771e0f695/src/modules/land_detector/FixedwingLandDetector.cpp#L78-L80

Now as a solution the first thing that comes to mind is just respecting the validity flag for horizontal velocity in the fixed-wing land detector but we need to consider what happens if the velocity is invalid. In the current way the land detector is written the horizontal velocity filter just doesn't get updated if the velocity is not valid and that could work if it was never valid because the filter stays zero and that passes the threshold. But in the more rare case where the estimate was ever valid (like e.g. in my simulation log) the problem would persist and get the filter stuck above the threshold. If we just make the check pass whenever the estimate is not valid that could potentially potentially lead to false positives.

My suggestion is to:
- keep the disarming behavior in manual thrust modes consistent across fixed-wing and multicopter for the same safety reasons
- revise the fixed wing land detector to consider thrust level and rotational movement such that the horizontal movement check can be skipped when the estimate is invalid without any significant risk of false positives.

### Software Version

main c024ea396a6065e1281b82877b0b7ba5f94aa41d

### Flight controller

pixhawk 4 mini

### Vehicle type

Fixed Wing

### Additional context

[image]
[image]

sfuhrer, 2024-03-25:
Thanks for the detailed write up!

> keep the disarming behavior in manual thrust modes consistent across fixed-wing and multicopter for the same safety reasons

I'm not yet so sure we should allow disarming in manual thrust modes even if not landed. After all, other than a MC, a FW can fly completely fine without thrust for a long while.

>revise the fixed wing land detector to consider thrust level and rotational movement such that the horizontal movement check can be skipped when the estimate is invalid without any significant risk of false positives.

Yes that one we clearly have to improve, and we really should share as much between the MC and FW land detector as possible :+1:

MaEtUgR, 2024-03-25:
@sfuhrer Thanks for having a look and your feedback!

> FW can fly completely fine without thrust for a long while, so having the stick at 0 doesn't mean we're landed as it generally the case for MC

Yes on the gesture you're totally right, it's not safe at all on fixed-wing manual mode. I was thinking in terms of my arm/disarm switch which I'd not use in air but would give me a reliable way to turn it off compared to adding a kill-switch that I would then use as a disarm switch but doesn't allow arming again of course.

MaEtUgR, 2024-04-13:
I started drafting land detector improvements on: https://github.com/PX4/PX4-Autopilot/tree/maetugr/fixed-wing-land-detector-improvements
Test logs of this version are here: https://github.com/PX4/PX4-Autopilot/pull/22668#issuecomment-2053759365

MaEtUgR, 2025-04-25:
If I'm not mistaken the underlying issue was addressed in https://github.com/PX4/PX4-Autopilot/pull/24133 and https://github.com/PX4/PX4-Autopilot/pull/24730. I haven't tested but otherwise I would reopen 👌

I should still bring in some of the improvements I started with in https://github.com/PX4/PX4-Autopilot/compare/main...maetugr/fixed-wing-land-detector-improvements

closed 2025-04-25 by MaEtUgR