PX4/PX4-Autopilot #22944: [Bug] Random land detection mid air, freefall detect not working

FrankvVeelen, 2024-03-28:
### Describe the bug

When testing our drone, we had a random land detect mid-air. When analysing we found out our ```MPC_THR_MIN``` is very close to our hover thrust estimate (25% vs ~29%). We think this could result in the random landing being detected.
However, after this a takeoff was detect (weird), but the motors didn't turn on again. 
Freefall was also not detected. When looking at the land detection module we found that for a freefall to be detected the norm of the acceleration should be >2.0. We feel that this is quite tight, especially in more windy conditions where some lift might still be generated.

### To Reproduce

Fly with ```MPC_THR_MIN``` set too high.

### Expected behavior

We expected that a random land detection would not result in a full crash, but freefall detect saving the drone.

### Flight Log

https://review.px4.io/plot_app?log=e2367fb6-f8de-46a3-b711-fcec82ce9968

### Software Version

1.14.0

### Flight controller

Holybro V6x

### Vehicle type

Multicopter

bresch, 2024-03-28:
Why did you have to set `MPC_THR_MIN` so high? Default value is 12%

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T19:51:48
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Random land detection mid air, freefall detect not working (Issue #22944)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.

FrankvVeelen, 2024-03-28:
> Why did you have to set `MPC_THR_MIN` so high? Default value is 12%

We had some issues where landings were not detected with the previous version of our drone. We then switched to higher kV motors, but didn't set the parameter back. We realize that this is part of the issue, but we still think freefall detect should've detected an issue

Seungwon-Yeo, 2024-03-28:
Dooray! 메일 발송 실패 안내

  
  
    
    
  
  
    
메일 발송
실패 안내
    
  
  
    
        
          
***@***.***) 
          님께
보낸
메일이
전송되지
못하였습니다.
        
        
          실패 사유를 확인해보세요.
        
    
  
  
    
      
        
          
            * 받는 사람 : 
***@***.***) 
          
          
            * 발송 시간 : 
2024-03-28T20:16:04
          
          
            * 메일 제목 : 
Re: [PX4/PX4-Autopilot] [Bug] Random land detection mid air, freefall detect not working (Issue #22944)
          
          
            
              
                * 실패 사유 : 
              
              
                
대외비/기밀 메일은 외부 계정으로 발송이 제한되어 일부 메일 발송 실패(수신자가 전달 설정한 경우 포함)
                
              
            
                      
          
        
      
    
  
  
    
        
          이 메일은 발신전용으로 회신되지 않습니다.
          더 궁금하신 사항은
          ***@***.***
          으로 문의해 주시기 바랍니다.
        
    
  
  
    
        © Dooray!.