PX4/PX4-Autopilot #22951: [Bug] There is no thrust_and_torque data type in the OffboardControlMode.msg message in version 1.14.2 of PX4.

Byzeal, 2024-03-29:
### Describe the bug

[image]

[image]

These two instructions do not correspond.

### To Reproduce

Customize thrust_and_torque in OffboardControlMode.msg

### Expected behavior

Whether future versions support it

### Flight Log

None

### Software Version

1.14.2

### Flight controller

PID

### Vehicle type

Multicopter

Jaeyoung-Lim, 2024-03-29:
@Byzeal If you enable actuator, this would allow you to command thrust and torque

Byzeal, 2024-03-29:
@Jaeyoung-Lim Thank you very much for your reply!   I want to do is to design control algorithm through Simulink and verify my algorithm through automatic code generation of PX4 firmware. What I do now is to enable actuator in the algorithm, does it mean that the official controller output of PX4 is shielded, so that the thrust and torque messages I publish can be verified?

Jaeyoung-Lim, 2024-03-29:
@Byzeal I do not know how the simulink is integrated with PX4, and with actuator setpoints. You would need to either verify this yourself or get support from mathworks.

Byzeal, 2024-03-30:
[image]
@Jaeyoung-Lim     Thank you very much for your reply! As shown in the image above, the model I built was generated using the Pixhawk toolchain with.pX4 firmware and burned into flight control. If not published, offboard_control_mode.msg message, I will be PX4 attitude loop output code after the comment out, re-generated code after burning can also take off and control normally.

closed 2026-04-08 by github-actions[bot]