PX4/PX4-Autopilot #22963: NXP Canhubk3 pwm_out driver missing and no dynamic control allocation in QGC [Bug]

dirksavage88, 2024-04-02:
### Describe the bug

The driver for pwm_out is missing in the [fmu board config](https://github.com/PX4/PX4-Autopilot/blob/main/boards/nxp/mr-canhubk3/fmu.px4board)

I can map ca_rotor params and choose an airframe, however when I look at the motors tab in QGC it is the old slider style and I do not see the control allocator tab. When I try to enable the motor sliders QGC reports that the command is not supported. 

### To Reproduce

1. Clone repo and build target 'nxp_mr-canhubk3_fmu'
2. upload to target
3. Open QGC and try to access the motors/actuators tab
4. actuators does not show dynamic control allocator tab, but old motor sliders
5. old motor sliders tab errors out

### Expected behavior

After choosing an airframe the control allocator screen should populate in QGC

### Flight Log

none

### Software Version

main

### Flight controller

nxp canhubk3 (with adap board)

### Vehicle type

Multicopter

dirksavage88, 2024-04-02:
[image]

dirksavage88, 2024-04-10:
@PetervdPerk-NXP was able to get this working, so it may be a board or SD card issue on my end. When boot for the first time with firmware default params no PWM is detected. When enabling PWM_MAIN_FUNCX the esc signal is detected and the correct tones are observed

dirksavage88, 2024-04-11:
`ERROR [PX4_MTD] failed to initialize EEPROM driver
ERROR [PX4_MTD] mtd failure: -5 bus 1 address 80 class 3
HW arch: NXP_MR_CANHUBK3
HW type: MR-CANHUBK3-ADAP
HW version: 0x000
HW revision: 0x000
PX4 git-hash: f9c65cd4c6e5badeeafca90aa1ced27feace71bf
PX4 version: 1.15.0 0 (17760256)
PX4 git-branch: main
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 61958c53056b5c3fb828f18a097de57d9bed0fda
Build datetime: Apr 10 2024 14:28:40
Build uri: localhost
Build variant: fmu
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00086125211322051365ffffffffffffffff
MCU: S32K344, rev. 0
INFO  [param] selected parameter default file /fs/microsd/params
ERROR [param] open '/fs/microsd/params' failed (2)
ERROR [init] param import failed
INFO  [parameters] summary: 0/1916 (used/total)
INFO  [parameters] file: /fs/microsd/params
INFO  [parameters] storage array: 0/1916 elements (256 bytes total)
INFO  [parameters] storage array (custom defaults): 0/1916 elements (256 bytes total)
INFO  [parameters] auto save: on
param: export: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
param: find: 0 events
param: get: 0 events
param: set: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
dmesg [24:100]
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board defaults: /etc/init.d/rc.board_defaults`

dirksavage88, 2025-06-19:
> Do we know the root cause of the issue?

No, we do not. I've seen similar issues with connections over wifi timing out and the metadata never successfully transfers. In this case my best guess was that the serial uart buffers were really undersized on the canhubk3.

dakejahl, 2025-06-20:
Is there a way to reproduce on an H7 board? I tried with a telem radio and couldn't reproduce.

closed 2026-02-06 by dirksavage88