PX4/PX4-Autopilot #22981: [Bug] CRASH DUMP on Pixhawk6x

FlavioTonelli, 2024-04-05:
### Describe the bug

I discovered a crash dump on one of my Pixhawk 6X units running the stable 1.14 version of the firmware. Currently, I have no idea what caused it or how to reproduce the issue.
[fault_2024_03_20_15_30_23.log](https://github.com/PX4/PX4-Autopilot/files/14883162/fault_2024_03_20_15_30_23.log)

### To Reproduce

Can't reproduce right now

### Expected behavior

Avoid crash of firmware

### Flight Log

dump log provided in the description

### Software Version

1.14 stable 

### Flight controller

Pixhawk 6x

### Vehicle type

None

FlavioTonelli, 2024-04-05:
Probably relates to #22481

federicociresola, 2024-04-10:
I had the same problem with the same hardware