PX4/PX4-Autopilot #22994: [Bug] Avoidance system not ready

enddrone, 2024-04-09:
### Describe the bug

not able to arm the drone because the avoidance system is not ready

### To Reproduce

enabled "COM_OBS_AVOID' and configured ''[CP_DIST], [CP_DELAY],[CP_GUIDE_ANG],[CP_GO_NO_DATA],[MPC_POS_MODE],[EKF2_EV_CTRL],
[EKF2_HGT_REF], [EKF2_EV_DELAY], [EKF2_EV_POS_X][EKF2_EV_POS_Y], [EKF2_EV_POS_Z], 
[image]
[image]

### Expected behavior

I want to use avoidance system using ros2 humble since it is the officially supported ros version for px4. What configuration should i use if i don't want to use the old px4 avoidance as shown in this repo (https://github.com/PX4/PX4-Avoidance.git) rather i want to mimic the same behavior using  my ros2 packages which publishes to vehicle_visual_odometry and obstacle_distance.

### Screenshot / Media

[image]
done all the needfull but still stuck with ''avoidance system not ready''

### Flight Log

dont have any logs

### Software Version

HW arch: CUBEPILOT_CUBEORANGEPLUS
PX4 git-hash: fa1885af238e94c089941d789f514901f0a8562e
PX4 version: 1.15.0 40 (17760320)
PX4 git-branch: main
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 693b9e782535f12e6a4ab657c7a0c3bd92b45fb1
Build datetime: Mar 27 2024 03:14:31
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000333838393032511200450046
MCU: STM32H7[4|5]xxx, rev. V

### Flight controller

Pixhawx 6C mini

### Vehicle type

Multicopter

xdwgood, 2024-04-10:
Can you confirm that vehicle_visual_odometry and object_distance have been accepted by the drone? (View using listener uorb topic)
Additionally, you need to send the advanced obstacle  system status in the heartbeat packet message.https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/mavlink/mavlink_receiver.cpp#L2179

hamishwillee, 2024-08-07:
This seems to be a common problem - see https://discuss.px4.io/t/using-collision-prevention/40013

hamishwillee, 2024-08-07:
There was a post here https://github.com/PX4/PX4-Avoidance/issues/568 that seemed to indicate that if you're using an external system to supply the messages then PX4 might require that you have a companion computer running and sending heartbeats/OK status. If so that isn't documented. I would be surprised because as long as you're getting the messages streamed at the right rate you could assume that the companion is OK.

@Jaeyoung-Lim Do you know what might cause this? Do we have an expert we can ask? I'm seeing this pop up from time to time, currently in https://discuss.px4.io/t/using-collision-prevention/40013

serkanMzlm, 2024-08-07:
> vehicle_visual_odometry ve object_distance'ın drone tarafından kabul edildiğini doğrulayabilir misiniz? (uorb konusunu dinleyen kişiyi kullanarak görüntüleyin) Ek olarak, gelişmiş engel sistemi durumunu kalp atışı paket mesajında ​​göndermeniz gerekir. https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/mavlink/mavlink_receiver.cpp#L2179

Is there any working ROS2 code in this section?

hamishwillee, 2024-08-07:
@serkanMzlm What section? AFAIK there is no working test code of PX4 collision detection or obstacle avoidance. The PX4-Avoidance repo is ROS1 "abandonware". 

The collision prevention code can in theory be tested without ROS or some other external source of obstacle information. The obstacle avoidance system needs an external source.

serkanMzlm, 2024-08-07:
@hamishwillee , my goal is to adapt the PX4-Avoidance repository to be compatible with ROS2 and Gazebo Garden. I'm working on it, but there are parts of the code where I'm stuck. Is there any documentation available to better understand the code?

Or, are there any avoidance-related settings in the PX4 configuration? Instead of converting the PX4-Avoidance repository directly, is there a feature in the PX4-autopilot software to perform obstacle avoidance using distance data?

hamishwillee, 2024-08-08:
@serkanMzlm

The documentation for obstacle avoidance, or more precisely "Obstacle Avoidance in Missions" is here: [Obstacle Avoidance](https://docs.px4.io/main/en/computer_vision/obstacle_avoidance.html). It indicates that all you need to do is set [COM_OBS_AVOID](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#COM_OBS_AVOID) to 1.

However as I noted in https://github.com/PX4/PX4-Autopilot/issues/22994#issuecomment-2272397660 I think there is also a dependency in PX4 on having a healthy companion computer. By implication, that means you need a component that is sending a mavlink `HEARTBEAT` with `type` [MAV_TYPE_ONBOARD_CONTROLLER ](https://mavlink.io/en/messages/common.html#MAV_TYPE_ONBOARD_CONTROLLER).

This is a bit unnecessary IMO. PX4 knows if it has a valid companion if it gets a stream of updates mirrored back. That is what I'd try though.

There is no way to use distance data for this at the moment.  I don't know what people might be planning to develop in future.

Good luck. Getting the avoidance repo working with ROS2 and regularly tested would be amazing.

hamishwillee, 2024-08-08:
PS And I see that the [planning interface](https://docs.px4.io/main/en/computer_vision/path_planning_interface.html#path-planning-interface] explicitly does require proof of life. To make it a bit easier for people to work out what is going on have added https://github.com/PX4/PX4-user_guide/pull/3340

closed 2024-08-08 by hamishwillee

serkanMzlm, 2024-08-08:
> @serkanMzlm
> 
> The documentation for obstacle avoidance, or more precisely "Obstacle Avoidance in Missions" is here: [Obstacle Avoidance](https://docs.px4.io/main/en/computer_vision/obstacle_avoidance.html). It indicates that all you need to do is set [COM_OBS_AVOID](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#COM_OBS_AVOID) to 1.
> 
> However as I noted in [#22994 (comment)](https://github.com/PX4/PX4-Autopilot/issues/22994#issuecomment-2272397660) I think there is also a dependency in PX4 on having a healthy companion computer. By implication, that means you need a component that is sending a mavlink `HEARTBEAT` with `type` [MAV_TYPE_ONBOARD_CONTROLLER ](https://mavlink.io/en/messages/common.html#MAV_TYPE_ONBOARD_CONTROLLER).
> 
> This is a bit unnecessary IMO. PX4 knows if it has a valid companion if it gets a stream of updates mirrored back. That is what I'd try though.
> 
> There is no way to use distance data for this at the moment. I don't know what people might be planning to develop in future.
> 
> Good luck. Getting the avoidance repo working with ROS2 and regularly tested would be amazing.

Thank you for the information. I will try and test 'MAVLink HEARTBEAT of type MAV_TYPE_ONBOARD_CONTROLLER' in the simulation environment