PX4/PX4-Autopilot #23005: [Bug] Offboard without GPS, 1.14

Pilwoo, 2024-04-11:
### Describe the bug

Hi, I'm trying to run offboard control without GPS in quadcopter

I'm using modifed example of 1.14 version with uxrce and ros2.

In QGC, it says no offboard signal and not ready to arm, but offboard signal is sent and checked from Mavlink console(urob top&listener)

Error : The offboard component is not sending setpoints or the required estimate (e.g. position) is missing. 

I think this problem is continued from Benjamin's report
https://github.com/PX4/PX4-Autopilot/issues/22456#issue-2016308804

As same as Benjamin, I changed the offboard mode code from 'Velocity' to 'Attitude', and offboard arming was success.

But what I want is control the drone to go forward 2 seconds (or 1 meter) and change the pitch (or reduce behind motor torque).

For now, I tried to send fake gps or fake global position for control the position or velocity.

These are what I tried and failed.

1. /fmu/in/vehicle_visual_odometry 

For the first, I tried to give visual odometry like Vicon. The data is sent well and can be checked in QGC uorb.

But still cannot arm...

2. /fmu/in/vehicle_global_position
actually /fmu/in/vehicle_global_position is not declared in dds_topics.yaml

When I sent the message, It doesn't shown in uorb topic list.

3.  /fmu/in/aux_global_position

this message is declared in dds_topic.yaml but not shown in uorb topic list.

4. /fmu/in/actuator_motors

I try to change the motor pwm input directly (i know this is dangerous) to fine adjust input for my mission.

I sent [1.1,0,0,~~~] to pixhawk, and I can find it sent well in my companion computor.

but in QGC, the sent message was different and shown the real actuator value as [0, 0.002, 0, 0.002, nan,nan, ~~~]

I don't know why it don't get the topic I sent.

------------------------------------------------------------------------

So for now, I don't know how to make the drone flying in non GPS... Thank you for reading

### To Reproduce

1. Drone switched on
2. connect manual RC
3. send offboard signal
4. error occur

### Expected behavior

motor should be armed and further it can fly.

### Screenshot / Media

[image]

[image]

[image]

### Flight Log

.

### Software Version

PX4 : 1.14

### Flight controller

Holybro pix32 v6

### Vehicle type

Multicopter

Jaeyoung-Lim, 2024-04-19:
@Pilwoo I think there are a few things to clarify.

Offboard control and the availability of GPS signal is not relevant. Offboard control is a mechanism to enable/disable different cascades of control loops from an external system. Therefore, the configuration of sensors for estimation is not relevant for enabling disabling offboard control.

On the other hand, this also means that if you are trying to use global position setpoints, you would need a valid position estimate.

In order to obtain a valid position estimate, you can use various sensors such as GPS or external vision systems such as Vicon or VIO. If there is no valid position estimate, you will not be able to fly any position related flight modes(such as Position), not just position setpoints with offboard.

You mention that you tried sending fake GPS data, but this cannot provide a valid position estimate, since your GPS data from fake_gps would be inconsistent with the other sensor data (such as IMU) which EKF will reject the GPS sensor data from the fake_gps module.

You also mention that you tried streaming vicon data, but it seems that there is something wrong on how the data is sent. I recommend you look carefully into the data frames, and quality of the data as well as whether the vicon data is getting correctly fused into the EKF2 data.

Hope this helps.

Pilwoo, 2024-04-20:
Thank you for your reply.

I have some question that

1. So, if I don't have GPS data or vicon data, is it impossible to use code that moves the vehicle forward even briefly?

2. Currently, GPS or vicon cannot be used because the vehicle must be driven indoors. If I want to give location information, what topic should I give it to?

and sorry for the labels, I'm not familiar to issue board. thank you

Jaeyoung-Lim, 2024-04-20:
> So, if I don't have GPS data or vicon data, is it impossible to use code that moves the vehicle forward even briefly?

The brievity of the motion is irrelevant. You cant move forward, because the vehicle doesnt know which direction forward is. You can still use lower level topics where the estimate is available (e.g. attitude, bodyrate etc)

> Currently, GPS or vicon cannot be used because the vehicle must be driven indoors. If I want to give location information, what topic should I give it to?

https://docs.px4.io/main/en/advanced/computer_vision

Pilwoo, 2024-04-21:
Thank you for your comment. I'll try and study harder.

closed 2024-04-21 by Pilwoo

albertocastrovill, 2024-10-11:
Hi @Pilwoo , I was wondering if you were able to solve the problem and test it in the scenario you wanted, move it 1 meter forward. If so, could you share your solution? Thnks