PX4/PX4-Autopilot #23013: Neither Gazebo Classic no jmavsim will work on Mac M1 arm

DonLakeFlyer, 2024-04-12:
### Describe the bug

I followed all the setup instructions including using an iTerm install which is running in Rosetta. This leaves me pretty hosed with respect to doing QGC testing.

Gazebo classic:
```
don@Donalds-Mac-Pro PX4-Autopilot % brew install px4-sim-gazebo
==> Downloading https://formulae.brew.sh/api/formula.jws.json

==> Downloading https://formulae.brew.sh/api/cask.jws.json

==> Fetching dependencies for px4/px4/px4-sim-gazebo: osrf/simulation/gazebo11
==> Fetching osrf/simulation/gazebo11
==> Downloading https://github.com/gazebosim/gazebo-classic/commit/3a9efee7ccdf552cbf22188131782f242f6c0542.patch?full_index=1
Already downloaded: /Users/don/Library/Caches/Homebrew/downloads/740e15a70c717de3cfc904342743e25cd1d228034f3573d51f23d6ad6093e0be--3a9efee7ccdf552cbf22188131782f242f6c0542.patch
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/don/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-11.14.0.tar.bz2
Already downloaded: /Users/don/Library/Caches/Homebrew/downloads/886d578ff7aa32ac5da8b63cc2ff79f24be5c9aff5092de102f6c6fd242f8b3d--gazebo-11.14.0.tar.bz2
==> Fetching px4/px4/px4-sim-gazebo
==> Downloading https://raw.githubusercontent.com/PX4/PX4-Autopilot/master/Tools/px4.py
Already downloaded: /Users/don/Library/Caches/Homebrew/downloads/be4aff9b473b8b0c6bae0ff08a6611885de9d56ee3a477c05453bb6bbcaab246--px4.py
==> Installing px4-sim-gazebo from px4/px4
==> Installing dependencies for px4/px4/px4-sim-gazebo: osrf/simulation/gazebo11
==> Installing px4/px4/px4-sim-gazebo dependency: osrf/simulation/gazebo11
==> Patching
==> Applying 3a9efee7ccdf552cbf22188131782f242f6c0542.patch
patching file 'gazebo/gui/qgv/QGVSubGraph.cpp'
==> Applying brew_python_fix.patch
patching file 'deps/opende/CMakeLists.txt'
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/opt/homebrew/opt/qwt-qt5/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/opt/homebrew/opt/qwt-qt5/lib/qwt.framework -DCMAKE_INSTALL_RPATH=@loader_path/../lib
==> make install
Last 15 lines from /Users/don/Library/Logs/Homebrew/gazebo11/02.make:
cd /tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build/deps/opende && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.25 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -DSPDLOG_FMT_EXTERNAL -DTINYXML2_IMPORT -D_FILE_OFFSET_BITS=64 -Dgazebo_ode_EXPORTS -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/test/gtest/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0 -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build/deps/opende -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/ou/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/OPCODE -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/GIMPACT/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/threadpool -I/opt/homebrew/Cellar/tinyxml2/10.0.0/include -I/opt/homebrew/Cellar/ffmpeg/7.0/include -I/opt/homebrew/include/ignition/msgs5 -I/opt/homebrew/include/ignition/transport8 -I/opt/homebrew/include/ignition/common3 -I/opt/homebrew/include/ignition/fuel_tools4 -I/opt/homebrew/Cellar/libccd/2.1_1/include -I/opt/homebrew/Cellar/bullet/3.25/include/bullet -I/opt/homebrew/Cellar/bullet/3.25/include -isystem /opt/homebrew/include/ignition/math6 -isystem /opt/homebrew/include/eigen3 -isystem /opt/homebrew/include/bullet -isystem /opt/homebrew/lib/urdfdom/cmake/../../../include/urdfdom -isystem /opt/homebrew/lib/urdfdom_headers/cmake/../../../include/urdfdom_headers -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic  -DNDEBUG -DdNODEBUG -DdDOUBLE -DHAVE_CONFIG_H -DPIC  -s -O3 -DNDEBUG  -std=c++17 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -fPIC -MD -MT deps/opende/CMakeFiles/gazebo_ode.dir/src/joints/dhinge.cpp.o -MF CMakeFiles/gazebo_ode.dir/src/joints/dhinge.cpp.o.d -o CMakeFiles/gazebo_ode.dir/src/joints/dhinge.cpp.o -c /tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/src/joints/dhinge.cpp
[ 14%] Building CXX object deps/opende/CMakeFiles/gazebo_ode.dir/src/joints/dball.cpp.o
cd /tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build/deps/opende && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.25 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -DSPDLOG_FMT_EXTERNAL -DTINYXML2_IMPORT -D_FILE_OFFSET_BITS=64 -Dgazebo_ode_EXPORTS -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/test/gtest/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0 -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build/deps/opende -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/ou/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/OPCODE -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/GIMPACT/include -I/tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/threadpool -I/opt/homebrew/Cellar/tinyxml2/10.0.0/include -I/opt/homebrew/Cellar/ffmpeg/7.0/include -I/opt/homebrew/include/ignition/msgs5 -I/opt/homebrew/include/ignition/transport8 -I/opt/homebrew/include/ignition/common3 -I/opt/homebrew/include/ignition/fuel_tools4 -I/opt/homebrew/Cellar/libccd/2.1_1/include -I/opt/homebrew/Cellar/bullet/3.25/include/bullet -I/opt/homebrew/Cellar/bullet/3.25/include -isystem /opt/homebrew/include/ignition/math6 -isystem /opt/homebrew/include/eigen3 -isystem /opt/homebrew/include/bullet -isystem /opt/homebrew/lib/urdfdom/cmake/../../../include/urdfdom -isystem /opt/homebrew/lib/urdfdom_headers/cmake/../../../include/urdfdom_headers -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic  -DNDEBUG -DdNODEBUG -DdDOUBLE -DHAVE_CONFIG_H -DPIC  -s -O3 -DNDEBUG  -std=c++17 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -fPIC -MD -MT deps/opende/CMakeFiles/gazebo_ode.dir/src/joints/dball.cpp.o -MF CMakeFiles/gazebo_ode.dir/src/joints/dball.cpp.o.d -o CMakeFiles/gazebo_ode.dir/src/joints/dball.cpp.o -c /tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/deps/opende/src/joints/dball.cpp
[ 14%] Building CXX object deps/opende/CMakeFiles/gazebo_ode.dir/src/joints/gearbox.cpp.o
cd /tmp/gazebo11-20240412-34395-pqq5q/gazebo-11.14.0/build/deps/opende && /opt/homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DDART_ACTIVE_LOG_LEVEL=2 -DDART_HAVE_spdlog=1 -DFMT_HEADER_ONLY=1 -DFMT_SHARED -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.25 -DLIBBULLET_VERSION_GT_282 -DMAC_OS [... cut ...]
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
don@Donalds-Mac-Pro PX4-Autopilot % make -j24 px4_sitl_default jmavsim
[1/2] launching px4 jmavsim_iris (SYS_AUTOSTART=10017)
[... 83 lines cut ...]
```

### To Reproduce

Steps described above...

### Expected behavior

That I can use either gazebo classic or jmavsim. At some point I did have gazebo classic working, but then I updated to latest main and it all broke.

### Flight Log

None

### Software Version

...

### Flight controller

...

### Vehicle type

None

DonLakeFlyer, 2024-04-13:
I was able to make progress by using a Linux ARM VM and running site there in Ubuntu 22. But even there it didn't work out of the box. I needed to do this to get gazebo to not crash:
```
pip uninstall empy
pip install empy==3.3.4
```

mengchaoheng, 2024-04-14:
the same problem

DonLakeFlyer, 2024-04-14:
So I forgot to say this was on latest Stable. I tried main on another machine and Gazebo worked without changes for me. Not sure about jmavsim. But at least for now I have a way forward with testing. @mengchaoheng You problem is different from mine. You should create a new issue.

closed 2024-04-14 by DonLakeFlyer