PX4/PX4-Autopilot #23026: [Bug]

yuxuanwu1234567, 2024-04-18:
### Describe the bug

I used motion capture data, not GPS, and when connecting with Mavros, the aircraft logo was not displayed on the QGC, even though I added the following code:
mavlink::common::msg::SET_GPS_GLOBAL_ORIGIN gpo;
        gpo.target_system = m_uas->get_tgt_system();
        gpo.latitude = 37.2343 * 1E7;
        gpo.longitude = -115.8067 * 1E7;
        gpo.altitude = 610.0 * 1E3; // roughly 610m above sea level
		
        UAS_FCU(m_uas)->send_message_ignore_drop(gpo);

[image]

### To Reproduce

.

### Expected behavior

.

### Screenshot / Media

.

### Flight Log

.

### Software Version

.

### Flight controller

pixhawk2.4.8

### Vehicle type

None

### How are the different components wired up (including port information)

.

### Additional context

.

yuxuanwu1234567, 2024-04-18:
@Jaeyoung-Lim

yuxuanwu1234567, 2024-04-22:
@Julian Oes

yuxuanwu1234567, 2024-04-22:
@julianoes

julianoes, 2024-04-23:
You didn't really fill out the issue template, so there is not enough information for us to ask your question. Plus you didn't actually give some context or ask a question or properly describe your problem.

Also, you started rudely tagging developers, so I'm just going to close this.

closed 2024-04-23 by julianoes