PX4/PX4-Autopilot #23066: [Bug] , [Stale]

luca0montecchio, 2024-04-28:
### Describe the bug

Hello everybody,

is there the possibility to perform a test flight from a controller I deployed in matlab/simulink?
I am currently trying but after I do the build, deploy and start in matlab and check that everything is as it should  I cannot procede with my objective.

### To Reproduce

The passages I follow are:

1. Set HITL mode on QGC with the HITL airframe
2. Go to matlab to do the hardware setup of the UAV Toolbox Support Package for PX4 Autopilots that in the new verion requires to use PX4-Autopilots v1.14.0
3. Perform monitor and tune on jmavsim and QGC through mavlink
4. Perform build, deploy and start in order to write the code in C++ inside the px4_simulink_app folder that is created after the monitor and tune from PX4-Autopilots/src/modules
5. I run the command `make px4_fmu-v3_default upload` in order to upload the firmware with the code on the board (the board is a cube-black that runs v3)
6. Head back to QGC in order to change the mode into `HITL and SIL disabled` and the airframe into a quadcopter

### Expected behavior

After the above passages and once all the sensors calibrations are completed it should be enough to bring the drone outside and make it fly following the trajectory I set in simulink and that I saw it following in the simulation.
When I attach the battery what I experience is nothing at all, the drone doesn't move and doesn't arm anymore not even witht the RC.

### Flight Log

 I cannot provide log file since it does not move.

### Software Version

v1.14.0

### Flight controller

PID

### Vehicle type

None

FilBk, 2024-05-09:
Unfortunatelly i'm not giving any proper solution. if would ask i from Qgc>Vehicle setup>actuators you can spin engines since it works by a different flow of as far as is in my understanding.

Moreover are you able to run the hitl simulation properly? I'm struggling with this too, and i may be wrong, but i'm suspecting some bug in the simulink app module within px4 1.14

luca0montecchio, 2024-05-29:
Hi, sorry for the late answer.
Yes I am able to run the simulation in HITL with PX4 v1.14.0 on Matlab 2024a. I am using Ubuntu 20.04, on windows you have to use the WSL2 that may complicate a bit the whole procedure.

FilBk, 2024-05-29:
Hi, In the meanwhile i fixed my problem for HITL simulation.

 I'm working ok Ubuntu 22, soon i will work to deploy of firmware for flight testing and i will be facing your same problem. 
  
  Still new to the environment but  i would ask , which topic  does  the firmware designed in simulink publish to?
 If it's pubblishing to HIL_ACTUATOR_CONTROLS,  as many examples in simulink are doing, i doubt it actually may spin the props even disabling the HIL mode from QGC.