PX4/PX4-Autopilot #23074: Enable Pixhawk to Receive MAVLink v2 Framed Messages [Bug]

anand2532, 2024-04-30:
### Describe the bug

I’m currently working on a project where I need to enable Pixhawk to receive MAVLink v2 framed messages. However, I’m encountering some difficulties in implementing this functionality. MAVLink v2 offers significant improvements over v1, especially in terms of efficiency and security, and enabling Pixhawk to handle v2 messages would be extremely beneficial for many projects.

The goal of this issue is to discuss and find a solution or workaround for enabling Pixhawk to receive MAVLink v2 framed messages

### To Reproduce

Newly Additional data of MAVLink V2 are not reflecting on QGC.

### Expected behavior

The goal of this issue is to discuss and find a solution or workaround for enabling Pixhawk to receive MAVLink v2 framed messages

### Flight Log

I just want the solution to sent the V2 framed messages.

### Software Version

version 14

### Flight controller

pixhwak 6x mini

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

We are sending the messages through TELEM2 port.

julianoes, 2024-04-30:
What do you mean with "framed messages"?

anand2532, 2024-05-02:
// Sending BATTERY_STATUS of an external BMS
msg := &common.MessageBatteryStatus{
    Id:               180,
    BatteryFunction:  common.MAV_BATTERY_FUNCTION_ALL,
    Type:             common.MAV_BATTERY_TYPE_LION,
    Temperature:      d.status.temperature,
    Voltages:         [10]uint16{c1, c2, c3, c4, c5, c6, c7, c8, c9, c10},
    CurrentBattery:   d.status.current,
    CurrentConsumed:  d.status.capacityConsumed,
    EnergyConsumed:   d.status.energyConsumed,
    BatteryRemaining: d.status.remaining,
    TimeRemaining:    d.status.timeRemaining,
    ChargeState:      d.status.chargeState,
    VoltagesExt:      [4]uint16{c11, c12, c13, c14},
    Mode:             d.status.mode_config,
    FaultBitmask:     d.status.bitmask,
}

As described in [MAVLink Common Messages](https://mavlink.io/en/messages/common.html):
TimeRemaining, ChargeState, VoltagesExt, Mode, and FaultBitmask are displayed in blue, indicating MAVLink 2 extension fields.

When sending these messages, MAVLink v1 messages are reflecting in QGroundControl, but MAVLink v2 extension fields are not reflecting. I inquired about this but didn't receive any response in [PX4 Discuss Forum](https://discuss.px4.io/) and [PX4 Discord](https://discord.gg/dronecode).

julianoes, 2024-05-03:
Can you link to the discuss post? And maybe explain a bit more what your problem is, maybe with some screenshots? I still don't understand what you mean.

anand2532, 2024-05-06:
Hey @JulianOes,

I have elaborated over the problem I have faced [here](https://discuss.px4.io/t/enable-pixhawk-to-receive-mavlink-v2-framed-messages/37985), thank you for the reply.

anand2532, 2024-05-08:
Hi @julianoes,

Can you please give me some direction on what might be the issue and how I can fix this?

anand2532, 2024-05-13:
Hey @julianoes ,
I have elaborated the problem with screen shot, please have a look [here](https://discuss.px4.io/t/enable-pixhawk-to-receive-mavlink-v2-framed-messages/37985)

mrpollo, 2024-05-14:
PX4 supports MAVLink 2, I'm closing this issue, and following up on forums

closed 2024-05-14 by mrpollo