PX4/PX4-Autopilot #23092: [Bug] Airspeed selector invalidating data

lashVN, 2024-05-03:
### Describe the bug

~10 minutes into flight, the airspeed selector is flagging the airspeed data as invalid

### To Reproduce

This issue doesn't occur every time, but is just 

### Expected behavior

Airspeed sensor not to be flagged as invalid

### Flight Log

[No failure occurs]https://logs.px4.io/plot_app?log=19d5eeee-54f5-461d-990e-f31572208cb4
[Midflight airspeed sensor failure](https://logs.px4.io/plot_app?log=a3002514-92a7-4156-acf5-420406a5567c)

### Software Version

Main build early May

c13e3bae

### Flight controller

Pixhawk 6X

### Vehicle type

None

### Additional context

System is using VectorNav driver instead of EKF2.

lashVN, 2024-05-03:
I've been walking through this log with @dagar, and we're having trouble discerning which check is causing the airspeed selector to invalidate the airspeed data. @sfuhrer , could you provide some context as to what might be triggering the failure flagging?

I'd especially be interested to ensure whether something in the VectorNav driver isn't publishing something as the airspeed selector is expecting.

Kim-zhi-jiang, 2024-05-07:
Found the reason？

sfuhrer, 2024-05-23:
The airspeed got disabled because the measurement was flat for more than 2s. 
[image]
Seems like the sensor you use has a low resolution of 5Pa, which then ofc makes this more likely to happen. 
You can disable the check by un-setting the second bit of ASPD_DO_CHECKS.

lashVN, 2024-05-23:
Hi @sfuhrer , thanks for your response. 

This issue has only been seen when using an external INS (not EKF2). Do you believe that is a coincidence then, that it has nothing to do with the faking on EKF2 consistency checks that we've run into in the past?

sfuhrer, 2024-05-24:
I assume it's a coincidence yes. 
What airspeed sensor are you using?

lashVN, 2024-05-28:
The airpseed sensor used is the [MS4525DO](https://holybro.com/products/digital-air-speed-sensor-ms4525do), which claims to have a resolution of 0.84 Pa. I'll see if there is something wrong in the configuration that may be causing it to output lower resolution.

In the more general case, how should I be debugging these failures? I don't seem to see any more specific message that provides the check that failed. I apologize if this is answered on a documentation page I've missed.

sfuhrer, 2024-05-29:
> The airpseed sensor used is the [MS4525DO](https://holybro.com/products/digital-air-speed-sensor-ms4525do), which claims to have a resolution of 0.84 Pa. I'll see if there is something wrong in the configuration that may be causing it to output lower resolution.

Could also be the driver in PX4 that publishes with that low resolution.

> In the more general case, how should I be debugging these failures? I don't seem to see any more specific message that provides the check that failed. I apologize if this is answered on a documentation page I've missed.

There is no documentation about it sadly. All the data you need is though logged (mostly check the topic airspeed_wind and airspeed_validated). And the main parameter that governs the check selection is ASPD_DO_CHECKS.

dk7xe, 2025-04-27:
Same issue it seems on my setup with I2C airspeed sensor it seems.
Maiden flight done
https://review.px4.io/plot_app?log=55f6fae4-0cef-47aa-bdcc-9f9639f70ae7
2nd flight
https://review.px4.io/plot_app?log=6c3d7a5b-98f7-496f-a5cb-5a8998ebd5d2

sfuhrer, 2025-04-28:
@dk7xe in your case the measured airspeed is negative, I don't see the connection to the other post in here. Was it working fine in previous flights, or have you changed the hardware since? You can often fix it by swapping the tubes of the pitot. 

[image]

dk7xe, 2025-04-28:
Shame on me, have acciddntly swapped the tubes 😢