PX4/PX4-Autopilot #23112: DroneCAN hardpoint support gripper hardpoint via RC channel

benjinne, 2024-05-09:
### Describe problem solved by the proposed feature

https://docs.px4.io/main/en/peripherals/gripper.html#supported-grippers

doesn't list CAN as a supported method for grippers to use, and I'm not seeing the function available.

My goal is to use the EPM v3 via uavcan as a gripper to control with an RC channel as documented in Ardupilot
https://ardupilot.org/copter/docs/common-electro-permanent-magnet-V3.html

It seems some initial support was added in: https://github.com/PX4/PX4-Autopilot/pull/19124
however it was never merged.

If anyone has suggestions on if this functionality already suggests or suggestions on how to implement, I'd be willing try and implement it.

### Describe your preferred solution

Implement DroneCAN hardpoint support via RC channel

### Describe possible alternatives

N/A

dirksavage88, 2024-05-09:
There is a [hardpoint controller](https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/uavcan/actuators/hardpoint.cpp) in the uavcan driver, but I am not sure how this would map to RC. A starting point is to be able to control the EPM via this hardpoint uavcan actuator message.

mrpollo, 2024-05-14:
it looks like this PR has been updated with the proper CAN message https://github.com/PX4/PX4-Autopilot/pull/19124