PX4/PX4-Autopilot #23114: [Bug] WARN  [health_and_arming_checks] Preflight Fail

kashifzr, 2024-05-09:
### Describe the bug

WARN  [health_and_arming_checks] Preflight Fail: vertical velocity unstable
WARN  [health_and_arming_checks] Preflight Fail: height estimate error
WARN  [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error

### To Reproduce

roslaunch px4 posix_sitl.launch vehicle:=iris_rplidar

### Expected behavior

Able to Run Rplidar with Iris without core fault or segmentation falt 

### Screenshot / Media

roslaunch px4 posix_sitl.launch vehicle:=iris_rplidar
... logging to /home/kkg/.ros/log/18de0df0-0dd0-11ef-b0ee-99ee7f070c17/roslaunch-kkg-W530-58961.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kkg-W530:44733/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    sitl (px4/px4)
    vehicle_spawn_kkg_W530_58961_3758604652260686476 (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [58976]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 18de0df0-0dd0-11ef-b0ee-99ee7f070c17
process[rosout-1]: started with pid [58993]
started core service [/rosout]
process[sitl-2]: started with pid [59000]

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
process[gazebo-3]: started with pid [59012]
INFO  [init] found model autostart file as SYS_AUTOSTART=1012
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 274 bytes, decoded 274 bytes (INT32:11, FLOAT:3)
process[gazebo_gui-4]: started with pid [59021]
INFO  [param] selected parameter backup file parameters_backup.bson
process[vehicle_spawn_kkg_W530_58961_3758604652260686476-5]: started with pid [59029]
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 39: [: Illegal number: 
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
[INFO] [1715237290.664198, 0.000000]: Loading model XML from file /home/kkg/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/iris_rplidar/iris_rplidar.sdf
[INFO] [1715237290.666219, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1715237291.002260003]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1715237291.003297535]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1715237291.192314571]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1715237291.193527064]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1715237291.478627078]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1715237291.500344808, 0.020000000]: Physics dynamic reconfigure ready.
[INFO] [1715237291.571868, 0.088000]: Calling service /gazebo/spawn_sdf_model
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
Warning [parser.cc:723] Can not find the XML attribute 'version' in sdf XML tag for model: rplidar. Please specify the SDF protocol supported in the model configuration file. The first sdf tag in the config file will be used 
Warning [parser.cc:833] XML Attribute[version] in element[sdf] not defined in SDF, ignoring.
[INFO] [1715237293.583409, 0.156000]: Spawn status: SpawnModel: Successfully spawned entity
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO  [lockstep_scheduler] setting initial absolute time to 168000 us
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-05-09/06_48_14.ulg	
INFO  [logger] Opened full log file: ./log/2024-05-09/06_48_14.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> [vehicle_spawn_kkg_W530_58961_3758604652260686476-5] process has finished cleanly
log file: /home/kkg/.ros/log/18de0df0-0dd0-11ef-b0ee-99ee7f070c17/vehicle_spawn_kkg_W530_58961_3758604652260686476-5*.log
INFO  [tone_alarm] home set
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [tone_alarm] notify negative
WARN  [health_and_arming_checks] Preflight Fail: vertical velocity unstable
WARN  [health_and_arming_checks] Preflight Fail: vertical velocity unstable
WARN  [health_and_arming_checks] Preflight Fail: vertical velocity unstable
WARN  [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: horizontal velocity unstable
WARN  [health_and_arming_checks] Preflight Fail: height estimate error
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error

### Flight Log

https://review.px4.io/plot_app?log=b022b0a0-74e2-4568-8210-450e7363b861

### Software Version

px4  : v1.14 
OS  :  Ubuntu Noetic 

### Flight controller

None 

### Vehicle type

Fixed Wing

mrpollo, 2024-05-14:
Hi @kashifzr I don't see a segmentation fault here, are you referring to the WARNINGS?

kashifzr, 2024-05-14:
@mrpollo when the estimation of vertical velocity , heading estimates are unstable then my drone without any command become unstable and start bouncing and run of space configuration. I try to overcome this issue by trying to look at so many options. But in the end when i remove models specific flow-cam and simple lidar  in iris_rplidar sdf file then my drone become stable,

mrpollo, 2024-05-16:
It looks like you found the problem!

kashifzr, 2024-05-16:
@mrpollo I did find the solution. when i used command make px4_sitl gazebo-classic_iris_rplidar without changes it run with out making issues, When i used command roslaunch px4 mavros_posix_sitl.launch vehicle:=iris_rplidar with out changes then same issues happened.

PHChenGit, 2024-05-22:
I also face the issue too. Any solutions?

kashifzr, 2024-05-22:
@PHChenGit can show where you find issue. its not related to rplidar or sdf file.

closed 2024-05-22 by kashifzr

chakloong883, 2025-01-09:
Hi! Have you resolve the problem? I meet the same problem.