PX4/PX4-Autopilot #23120: [Bug] __orb_trajectory_setpoint</b>’ was not declared in this scope

Crotonbee, 2024-05-11:
### Describe the bug

Hi，I have written a code with PX4 1.14 to do a simulation on gazebo,like this:  uORB::Publication<vehicle_local_position_setpoint_s>	_trajectory_setpoint_pub{ORB_ID(trajectory_setpoint)};

### To Reproduce

but it suported a mistake:</b></font>‘<b>__orb_trajectory_setpoint</b>’ was not declared in this scope; did you mean ‘<b>__orb_position_setpoint</b>’?
but if I run this code with 1.13,it works well

### Expected behavior

So how should I use trajectory_setpoint in 1.14?Thanks!

### Flight Log

<pre>In file included from <b>../../platforms/common/uORB/Subscription.hpp:41</b>,
from <b>../../src/modules/control_node/control_node.h:5</b>,
from <b>../../src/modules/control_node/control_node.cpp:1</b>:
<b>../../platforms/common/uORB/uORB.h:80:25:</b> <font color="#CC0000"><b>error: </b></font>‘<b>__orb_trajectory_setpoint</b>’ was not declared in this scope; did you mean ‘<b>__orb_position_setpoint</b>’?
80 | #define ORB_ID(_name)  &<font color="#CC0000"><b>_orb</b></font>##_name
|                         <font color="#CC0000"><b>^~</b></font>
<b>../../src/modules/control_node/control_node.h:83:86:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro ‘<b>ORB_ID</b>’
83 |        uORB::Publication<vehicle_local_position_setpoint_s> _trajectory_setpoint_pub{<font color="#06989A"><b>ORB_ID</b></font>(trajectory_setpoint)};
|                                                                                      <font color="#06989A"><b>^~</b></font>
compilation terminated due to -Wfatal-errors.
[3/471] Building CXX object src/module...es/modules__ekf2.dir/EKF/control.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_sitl_default] Error 1
</pre>

### Software Version

1.14

### Flight controller

Nxt-dual

### Vehicle type

None

hamishwillee, 2024-05-15:
Not an expert but `src/modules/control_node/control_node.h` does not appear to be part of PX4 code, so it is hard to know what it includes or does not include.

See if the following line is somewhere in your include path: `#include <uORB/topics/trajectory_setpoint.h>`

Crotonbee, 2024-05-15:
> Not an expert but `src/modules/control_node/control_node.h` does not appear to be part of PX4 code, so it is hard to know what it includes or does not include.
> 
> See if the following line is somewhere in your include path: `#include <uORB/topics/trajectory_setpoint.h>`

Thank you very much for your help! The problem has been solved

closed 2026-03-19 by github-actions[bot]