PX4/PX4-Autopilot #23124: [Bug] Cannot build

Midren, 2024-05-13:
### Describe the bug

https://github.com/Auterion/FlightGear-Rascal  repository is absent, but needed for the compilation

### To Reproduce

Build the px4:
```
$ git clone https://github.com/PX4/PX4-Autopilot /opt/px4 &&     cd /opt/px4 &&     git checkout v1.14.2 &&     make submodulesclean &&     make submodulesupdate &&     pip3 install --user kconfiglib empy pyros-genmsg jsonschema toml packaging future jinja2 &&     bash Tools/setup/ubuntu.sh --no-nuttx &&     DONT_RUN=1 make -j$(nproc) px4_sitl_default gazebo
```

See error:

```
16 59.73 fatal: could not read Username for 'https://github.com': No such device or address
#16 59.74 fatal: clone of 'https://github.com/Auterion/FlightGear-Rascal.git' into submodule path '/opt/px4/Tools/simulation/jsbsim/jsbsim_bridge/models/Rascal' failed
#16 59.74 Failed to clone 'models/Rascal'. Retry scheduled
#16 59.97 fatal: could not read Username for 'https://github.com': No such device or address
#16 59.97 fatal: clone of 'https://github.com/Auterion/FlightGear-Rascal.git' into submodule path '/opt/px4/Tools/simulation/jsbsim/jsbsim_bridge/models/Rascal' failed
#16 59.97 Failed to clone 'models/Rascal' a second time, aborting
```

Link to https://github.com/Auterion/FlightGear-Rascal also shows that repository does not exist

### Expected behavior

Build px4

### Flight Log

None

### Software Version

Tested on 1.14.2, 1.13.2

### Flight controller

SImulation

### Vehicle type

None

mrpollo, 2024-05-14:
@Midren I just tried building the v1.14.2 release and I don't see this problem, the submodule is pulling from `github.com/ThunderFly-aerospace/FlightGear-Rascal` not from any Auterion repository

@bkueng do you know if this is an issue?

MaEtUgR, 2024-05-15:
Thanks for the report! It is an issue. I just found this in #23126 and I think I have a hotfix that makes it available again now. But we should replace the submodule on PX4 main and possibly the last release to get rid of the dependency and avoid future problems.

Here's the tree of commits where the one from @RomanBapst for the electrical motor needs to stay available on the Auterion repo in the meantime.
[image]

I pushed this branch from my local clone: https://github.com/Auterion/FlightGear-Rascal/tree/preserve-this-branch-for-px4-submodules
And also updated the master branch of the repository to have these changes merged again.

MaEtUgR, 2024-05-15:
I verified it clones again completely on a fresh VM using just https without authentication.
[image]
Now we can remove the dependency on main to avoid any future problems.