PX4/PX4-Autopilot #23156: PX4 ROS2 Parameter Bridge

dakejahl, 2024-05-19:
### Describe problem solved by the proposed feature

With the introduction of the px4-ros2-interface-lib and the long term goal of tight integration between PX4 firmware and ROS2 linux, we need a way to unify the parameter systems of PX4 and ROS2. 

### Describe your preferred solution

uORB version of the mavlink parameter microservice?

### Describe possible alternatives

:shrug

### Additional context

One example is ROS2 mode which overrides Precision Land, you would want to reuse the existing PLD_ parameters. Users at the GCS shouldn't need to know or care where modes live.

dakejahl, 2024-05-19:
I think we need a new module **ParameterManager** which implements the parameter microservice via uORB. This would then be the common interface in PX4 for mavlink, uavcan, and dds. The parameter system really should be unified across all of these different protocols and subsystems.

RaidZeroSix, 2026-01-08:
It looks like the capability to resolve parameters by name (String -> Index) is currently siloed within the MAVLink module.

Although existing uORB messages like ParameterSetValueRequest exist, they rely on parameter_index. This works for the internal primary/remote system (which assumes synchronized firmware builds), but breaks for ROS2 since parameter indices change between builds.

I agree with @dakejahl that a dedicated ParameterManager module is needed to address that. It  would effectively need to expose a 'String-based Parameter Microservice' over uORB. That would allow DDS (and other protocols) to bridge these new request/response topics, letting external nodes query/set parameters by name without knowing the internal build-specific indices.