PX4/PX4-Autopilot #23157: EKF2 position estimates are way too off compared to precise RTK GPS location[Bug]

slgrobotics, 2024-05-19:
### Describe the bug

I have a dual RTK GPS setup on a rover (lawnmower) - delivering centimeter accuracy _Lat/Lon/Alt_ and _heading_ with sub-degree precision. This data is, naturally, fused by EKF2 and its estimates are published as _VehicleLocalPosition_ and _VehicleGlobalPostion_ messages. I started logging the discrepancy between the Global and RTK GPS data, and here is the typical result:
[image]
While the 0.3 meter deviation and oscillations may be acceptable for a drone in the sky, it completely messes up any ground base vehicle trying to stick to its course.

My solution to this problem was to override _Lon/Lat x/y_ and _heading_ values immediately after receiving the _*Position_ messages, and that works well.

I still would consider delivering imprecise estimated values when accurate location data is available to be a bug/oversight.

It would be prudent to pass through unmolested RTK GPS data in these fields when GPS _fix_type==6_ 
 

### To Reproduce

Have an RTK GPS equipped rover or drone and log discrepancy between actual coordinates and EKF2 estimates

### Expected behavior

EKF2 should fully use accurate RTK GPS data and not substitute it with imprecise estimates

### Flight Log

https://review.px4.io/plot_app?log=6be08335-a60f-468b-93cb-f16291ef5688

### Software Version

Current main

### Flight controller

Raspberry Pi 4

### Vehicle type

Rover

slgrobotics, 2024-05-19:
Just FYI - my code substituting the position and heading is at lines 160-190 here:

https://github.com/slgrobotics/PX4-Autopilot/blob/main/src/modules/rover_pos_control/RoverPositionControlPolling.cpp

xdwgood, 2024-05-20:
> EKF2 should fully use accurate RTK GPS data and not substitute it with imprecise estimates

I feel this can be achieved by adjusting the EKF parameters.eg：ekf2_gps_p_noise、ekf2_gps_v_noise

slgrobotics, 2024-05-20:
@xdwgood  - thanks for the hint, I will try changing the *NOISE parameters soon and get those deviations logged.

Meanwhile I think there's a philosophical question here - if you have EXACT ACCURATE measurements - what is the best strategy for Position Estimator to apply: pass them through or apply its usual imprecise math and hope for the best? BTW, the values which aren't available with precision should be still estimated and delivered as usual (accelerations, for example).

@bresch @dagar - any opinion?

YasinBedirhanSimsek, 2024-05-20:
I am also looking into this. Reducing the noise coefficients in sensor_gps_sim makes the Gazebo readings more true for X and Y.

X and Y positions get close to the actual gazebo coordinates but the Z axis is still not good enough for a rover or good enough for me.

Adjusting ekf2_gps_p_noise、ekf2_gps_v_noise also did not work very well but at it least just reduced the GPS vertical drift but it is still there. Even if I set the random GPS noise to 0 and just make the sensor_gps_sim module publish ground truth

https://github.com/PX4/PX4-Autopilot/blob/6b0ac49daf5c49aeb625ed684359fdd59b1a287d/src/modules/simulation/sensor_gps_sim/SensorGpsSim.cpp

slgrobotics, 2024-05-23:
Just to be fair to EKF2, it does what it is supposed to do - estimating/predicting position between GPS samples.

I've got RTK GPS running at 8HZ+, and decreased **EKF2_GPS_P_NOISE** to 0.05 and **EKF2_GPS_P_NOISE** to 0.1

Here is the same graph with speeds overlaid - when speed is higher, the predicted position diverts (forward?) more - from almost 0 to 0.2 m:

[image]

At 1.5 m/s speed and 8Hz GPS sampling rate the rover moves ~0.2 m between the samples.  

In my specific case it is easier for my Position Control code to use RTK GPS data directly, as it gets confused with the noise produced by EKF2. Your mileage may vary.

Flight log: https://review.px4.io/plot_app?log=3fa377f4-d2f2-40c6-a565-19ba4338433e

YasinBedirhanSimsek, 2024-05-23:
Gz bridge module takes the navsat sensor data and publishes it on the ground truth topic

SensorGpsSim module subscribes to the ground truth, adds noise to the measurements, and sets other GPS values like HDOP, VDOP, epv eph.

Then publishes the noise-added GPS data on the sensor GPS topic. Currently, the added noise looks like it is for a normal GPS but with good signal quality.  

It might be much better to change this module and add some parameters so that it can also simulate an RTK GPS. 

@dagar

YasinBedirhanSimsek, 2024-05-24:
ChatGPT says that the random noise generator function has a mean of 0 and a standard deviation of 1.

The randomly generated value is then scaled and added to the GPS data, however, 

I was able to change the X and Y accuracy but the Z accuracy is still not good enough (not good enough for me at least).

I can get 2cm accuracy for X and Y but Z keeps bouncing between +10cm and -10cm. So basically X and Y mean is 0 and the error margin is 2cm. Same for the Z but margin is 10cm and sometimes bigger depending on the parameters etc...  

Making the Z more accurate is possible but this causes X and Y to have an offset of 3-4 cm.  

[image]

slgrobotics, 2024-05-24:
@YasinBedirhanSimsek  - this issue is about real RTK GPS hardware and actual Rover, not a Gazebo SITL. Let's not confuse the two.

YasinBedirhanSimsek, 2024-05-24:
@slgrobotics, Sorry for that, I will open a new issue for SITL then. My bad

slgrobotics, 2024-05-24:
@YasinBedirhanSimsek  - no problem, I'd love to see GPS RTK simulated in Gazebo, that would be great help!

YasinBedirhanSimsek, 2024-05-24:
Besides EKF2_GPS_P_NOISE and EKF2_GPS_V_NOISE, you could try changing the EKF2_GPS_DELAY, EKF2_TAU_POS and EKF2_TAU_VEL

bresch, 2024-05-24:
@slgrobotics To get a good consistency between the inertial navigation (from IMU) and your high accuracy GNSS receiver, everything need to be precisely aligned. For example, you need to set the exact position of the GNSS antenna in `EKF2_GPS_POS_X/Y/Z` and the delay in `EKF2_GPS_DELAY`. If the delay parameter is not high enough, at each acceleration the EKF will first need to slow down (because it started to move before the GNSS velocity) and then re-accelerate to catch up with the lat/lon measurement; similar issue if the delay parameter is too high.

slgrobotics, 2024-05-27:
I tried EKF2_GPS_DELAY values 50 and 200, and neither showed improvement compared to default 110.

One peculiar feature of my lawnmower setup is a "controller head" positioned high over the center mass of the vehicle. It shakes a lot - less from the gas engine vibrations, more side-to-side because of uneven lawn surface. IMU is, naturally, in the center of that "head". The other (electrically driven) bot experiences less vibrations. EKF2 might be confused, among other things, by accelerometer input. Placing the controller box with sensors lower, where the seat is would somewhat diminish that effect.

Tractors and other agricultural machines don't have suspension and their seats are designed to only cushion large vertical jolts. Sudden lateral displacement is common and quite uncomfortable for humans in these vehicles.

[image]

bresch, 2024-06-03:
@slgrobotics Another param you should try to set as accurately as possible are the position of the IMU and GNSS antenna w.r.t the CG of the vehicle (no need to exactly be the CG but some fixed point). 
The parameters are: `EKF2_GPS_POS_X/Y/Z` and `EKF2_IMU_POS_X/Y/Z`. Those are important to compensate for the relative velocity between the IMU and GNSS when the vehicle is rotating.

slgrobotics, 2024-06-03:
@bresch - I already did that, even set GPS_PITCH_OFFSET to -45 (second antenna is in front at the feet level, related to primary on the head's top).

At the moment I've got the lawnmower follow the line with +-20 cm precision, and the EKF2/RTK discrepancy stays mostly within 10 cm - so my problems have been resolved. I still use 8 Hz RTK values directly for navigation, switching to EKF2 makes things worse.

I think that there's a lot of room for improvement if PX4 could be thoroughly tested on larger rovers with RTK GPS. This ticket could be closed or just kept open as a reminder.

slgrobotics, 2024-06-03:
@bresch - I see that EKF2 uses GNSS velocity readings to fuse with mag for heading. But dual antenna RTK GPS delivers heading directly. Is it being used in EKF2? If not, that's a huge omission...

bresch, 2024-06-03:
> I already did that

Ok, because in the log you shared [here](https://github.com/PX4/PX4-Autopilot/issues/23157#issuecomment-2128218554) those params were set to 0.

> I see that EKF2 uses GNSS velocity readings to fuse with mag for heading. But dual antenna RTK GPS delivers heading directly. Is it being used in EKF2?

The EKF uses GNSS heading when available (and when activated in `EKF2_GPS_CTRL`) instead of mag. I can see in your log that GNSS heading is being used (not mag).

bresch, 2024-06-04:
Ok, no worries.
The tricky thing with RTK is that every detail start to matter if you want to keep that high accuracy when mixing it with other sensors, otherwise you're just adding worse measurements to it, which leads to an overall worse accuracy.

I'll close the issue for now; we can still re-open if needed.

closed 2024-06-04 by bresch