PX4/PX4-Autopilot #23159: Drone is spinning/YAW (CW, CCW) randomly during flight [ARKFlow, PX1.14.0, Pixhawk 6C]

MuhammadBilal1, 2024-05-21:
### Describe the bug

During the flight, my drone is doing the **YAW** sometimes CW and other times CCW automatically without sending the YAW from the RC in the Position/Altitude flight.
It remains stable for a few seconds and then starts coming down and/or start spinnig.

**Here are the details of the setup:**
- Pixhawk 6C
- PX4 1.14.0
- ARKFlow
- No GPS
- Indoor flight
- Position/Altitude flight modes
- Default tuning parameters
- Custom airframe

**Note:** I got a [X500 V2 Kits](https://holybro.com/products/x500-v2-kits?variant=42541212041405) and that is working absolutely fine at the same place with the same [ARK Flow](https://arkelectron.com/product/ark-flow/) sensor.

### To Reproduce

Position/Altitude flight

### Expected behavior

Stable flight without spinning at yaw.

### Flight Log

https://review.px4.io/plot_app?log=f4965f81-ef97-4c65-8ac1-cdda33dc9103

### Software Version

PX4 1.14.0

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

xdwgood, 2024-05-22:
From the log, I suggest that you keep the mag sensor away from the power cord
[image]

AlexKlimaj, 2024-05-22:
https://review.px4.io/plot_app?log=f4965f81-ef97-4c65-8ac1-cdda33dc9103#Nav-Motor-Outputs

Something is also going on with your motors/props. Two of your motors are at 100% most of the flight.

MuhammadBilal1, 2024-05-22:
> https://review.px4.io/plot_app?log=f4965f81-ef97-4c65-8ac1-cdda33dc9103#Nav-Motor-Outputs
> 
> Something is also going on with your motors/props. Two of your motors are at 100% most of the flight.

Is this also possible that this is happening because of the drone being over weight?

MuhammadBilal1, 2024-05-22:
> From the log, I suggest that you keep the mag sensor away from the power cord ![Screenshot from 2024-05-22 10-12-30](https://private-user-images.githubusercontent.com/24932258/332632637-8d599da7-cee3-4fab-abcf-05842f89c1ed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzNTA2MDksIm5iZiI6MTcxNjM1MDMwOSwicGF0aCI6Ii8yNDkzMjI1OC8zMzI2MzI2MzctOGQ1OTlkYTctY2VlMy00ZmFiLWFiY2YtMDU4NDJmODljMWVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIyVDAzNTgyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyMzU0NzJjMTVhZmM4MmUzOTg2OWYyZmE4ZTBjOWNmNzFkM2JkYzI0NGZkOWMyYmJiMWE0NTUwNzM3MTMwODcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uVLzIUes8dM5TVTLOBWRuReJVu8QzqDd96W94Rb3irc)

Ok, I will adjust the position of the Flight controller.

julianoes, 2024-09-09:
Closing as there was no more follow up.

closed 2024-09-09 by julianoes