PX4/PX4-Autopilot #23164: takeoff issue[Bug]

airsimdevdd, 2024-05-22:
### Describe the bug

cannot take off the uav without RC even if i changed the mode to stick input disabled i cannot take of the uav and when the com_rc_in_mode is in RC transmitter only it is unable to take off(when i arm the uav first using RC and then give the take off command then it is taking off )  

### To Reproduce

When the uav is given the take off from the QGC, the drone gets armed and within some seconds it disarms saying Disarmed by auto preflight disarming 
1.  Drone switched on and uploaded the mission from QGC 
2. Mission is take off from the ground to 15m altitude and return to launch position
3. drone gets armed and went to take off mode and gets disarmed within 5 seconds but not taking off
4. error is disarmed by auto preflight disarming 
[params.txt](https://github.com/PX4/PX4-Autopilot/files/15398997/params.txt)

5. my parameters are 

### Expected behavior

When there is no RC nor joystick the uav needs to take off when the command is given from the QGC. here the QGC version is 4.2.9

### Screenshot / Media

[image]
[image]
[image]

### Flight Log

[params.txt](https://github.com/PX4/PX4-Autopilot/files/15399004/params.txt)

### Software Version

PX4 version : 1.14.0 

### Flight controller

cube orange 

### Vehicle type

Multicopter

airsimdevdd, 2024-05-22:
there is no rc loss failsafe trigger disable option

federicociresola, 2024-10-01:
I think the problem is that you have a big drift on GPS. Have you check the GPS information and the EKF2 give you the correct position?