PX4/PX4-Autopilot #23187: "[uavcan:125:vehicle_imu] Gyro 0 clipping, not safe to fly!" | "gyro 0 (2490C] 39.0 degC" [ARKFlow | PX4 1.14.0 | Quadcopter | No GPS ]

MuhammadBilal1, 2024-05-27:
### Describe the bug

Hi,
I am getting the issue of **"[uavcan:125:vehicle_imu] Gyro 0 clipping, not safe to fly!"** | **"gyro 0 (2490C] 39.0 degC"** randomly in different flights.
I found these messages in the "Show additional data" portion of the PX4 "Flight Review".
ARKFlow is mounted at the belly of the drone.

What could be the reason of this and what is the meaning of these messages?

**Here are the details of the setup:**

- Pixhawk 6C
- PX4 1.14.0
- ARKFlow
- No GPS
- Indoor flight
- Position/Altitude flight modes
- Self created airframe

**Problem:** Usually am getting the problem of drone not holding the position while using the ARKFlow.

### To Reproduce

Use the ARKFlow module.

### Expected behavior

Stable flight with drone holding the position indoor.

### Screenshot / Media

[image]

[image]

[image]

### Flight Log

**Flight Log # 1:** 1 minute flight: (Gyro 0 clipping, not safe to fly! | gyro 0 (2490C] 37.5 degC)
https://review.px4.io/plot_app?log=4e2db729-de17-4946-a805-1b9a96483216

**Flight Log # 2:** 54 seconds flight: (Gyro 0 clipping, not safe to fly! | gyro 0 (2490C] 39 degC)
https://review.px4.io/plot_app?log=0f4f591b-7c2d-4dc8-99e4-864106feb958

**Flight Log # 3:** 3 minutes 14 seconds flight: (Gyro 0 clipping, not safe to fly! | gyro 0 (2490C] 44.0 degC)
https://review.px4.io/plot_app?log=fc2fbd08-e9bd-463d-996a-6abb42799275

**Flight Log # 4:** 7 minutes 30 seconds (Gyro 0 clipping, not safe to fly! | gyro 0 (2490C] 39.1 degC)
https://review.px4.io/plot_app?log=de3e408d-0347-4c43-af55-f8aca5856e4f

### Software Version

PX4 1.14.0

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

ARKFlow is installed and configured in the drone at CAN port according to the [PX4 ARKFlow](https://docs.px4.io/main/en/dronecan/ark_flow.html) guide.

MuhammadBilal1, 2024-05-27:
@bresch I saw one of your comment [here](https://discuss.px4.io/t/accel-clipping/27659/9) about the "Accelerometer clipping".
Is my problem also similar to that?

@AlexKlimaj I believe this issue is related to the ARKFlow, can you please help me here!

AlexKlimaj, 2024-05-28:
Build and flash the latest main ARK Flow firmware. The clipping warning has been silenced.

MuhammadBilal1, 2024-05-28:
> Build and flash the latest main ARK Flow firmware. The clipping warning has been silenced.
> 
> #22441

There will be any impact on the flight due to this clipping or the temperature?

AlexKlimaj, 2024-05-28:
The temperature message is just telling you the temperature at which the auto calibration ran.

Gyro clipping isn't great on the ARK Flow since the gyro is used to compensate the optical flow data.

MuhammadBilal1, 2024-05-28:
> The temperature message is just telling you the temperature at which the auto calibration ran.
> 
> Gyro clipping isn't great on the ARK Flow since the gyro is used to compensate the optical flow data.

Ok. Thank you for the clarification.

But, I am continously losing the altitude during the hover and sometimes the loss of position as well.
Here are the logs of the autonomous flights:

**Flight log # 1:** (Offboard mode)
https://review.px4.io/plot_app?log=a54a7d4a-98ef-4560-9491-c6df4d19268c

**Flight log # 2:** (Offboard mode)
https://review.px4.io/plot_app?log=defb2b4a-4902-4705-9dbb-a8d7f1ff9c42

**Flight log # 3:** (Position mode)
https://review.px4.io/plot_app?log=77e8e481-8ac7-4aec-808c-7150bba2b82b

**Flight log # 4:** (Offboard mode)
https://review.px4.io/plot_app?log=fdf6593a-77e8-4d63-8855-72acb42fdefc

AlexKlimaj, 2024-05-28:
Try 1.15 beta on your flight controller. See MPC_ALT_MODE param.

MuhammadBilal1, 2024-05-29:
> Try 1.15 beta on your flight controller. See MPC_ALT_MODE param.

We tried the following, but the issue is still the same:
- Updated the firmware to PX4 1.15.0-beta1
- Updated the firmware of ARKFlow to the latest

Here are the logs:

**Flight log # 1**
https://review.px4.io/plot_app?log=9fcad401-68a5-48ad-bc00-6b973528314a

**Flight log # 2**
https://review.px4.io/plot_app?log=b89acfa9-e170-40bf-9873-38add4b232e5