PX4/PX4-Autopilot #23200: [Bug] CI on main fails

MaEtUgR, 2024-05-29:
### Describe the bug

1. 
[image]
[image]
fixed by https://github.com/PX4/PX4-Autopilot/pull/23192
2.
[image]
[image]
fixed by https://github.com/PX4/PX4-Autopilot/pull/23198
3.
[image]
[image]
fixed by https://github.com/PX4/PX4-Autopilot/pull/23197
4.
[image]
[image]
fixed by https://github.com/PX4/PX4-Autopilot/pull/23174
5. 
[image]
[image]
6.
[image]
[image]
Both heavy flash overflow. Any idea? @dagar ? 😇 
7. 
[image]
[image]
@sfuhrer started looking into it
8.
[image]
[image]
@PerFrivik started looking into it

### To Reproduce

Look at CI report on new commits on main.

### Expected behavior

Passing CI such that PRs see when they brake things.

### Flight Log

See screenshots

### Software Version

main branch

### Flight controller

CI

### Vehicle type

None

julianoes, 2024-05-29:
Thanks @MaEtUgR. I wonder if I should have been able to catch this before hitting the release.

MaEtUgR, 2024-05-30:
@julianoes have been... let's fix things and take learnings 👍 Thanks for all your efforts.

I think it's now down to flash overflow issues and possibly some (intermittent?) Jenkins problem. I'll keep this issue updated.

PetervdPerk-NXP, 2024-05-30:
Since #23088 FMU-V5 Stackcheck is overflowing
**https://github.com/PX4/PX4-Autopilot/actions/runs/9302858329/job/25603939618#step:7:10531**

MaEtUgR, 2024-06-03:
@PetervdPerk-NXP Sad story but it looks to me like https://github.com/PX4/PX4-Autopilot/pull/23169 freed up some flash again.
At least `main` passed the last run
[image]
Probably the VTOL test is intermittent since it failed on the pr that was already rebased just before that. But at least we're through the main issues. Let's keep it that way and fix the intermittent ones.