PX4/PX4-Autopilot #23237: [Bug]

serkanMzlm, 2024-06-06:
### Describe the bug

I added my own model and world to the PX4 Autopilot software. The model (Intel RealSense D435) works fine, but when I run it with the world model, the drone model does not load.

### Depth Camera D435i model: 
[image]

### World:
[image]

### Bug
[image]

### To Reproduce

1. I ran the PX4 Autopilot software in the simulation environment with my model on the default world model (it worked properly).
2. When I tested the x500 with my own world model, it did not work properly

[image]

### Expected behavior

I want PX4 to work properly in the world model I have prepared.

### Flight Log

In the simulation environment...

### Software Version

main branch

### Flight controller

x500

### Vehicle type

Multicopter

xdwgood, 2024-06-07:
@serkanMzlm  see here:https://github.com/PX4/PX4-gazebo-models/pull/44#discussion_r1630505818

The modified drone can display normally, but I have the following two mistakes:
 Unable to find file with URI [file://media/dem/data.dt2]
Unable to find file with URI [file://media/dem/data.dem]

I really didn't see data.dem under the dem file.Curiously, data.dt2 exists, but it cannot be find.

serkanMzlm, 2024-06-07:
@xdwgood  Yes, I corrected the part with data.dem. It should indeed be data.dt2

xdwgood, 2024-06-07:
[image]

try new,but not work

serkanMzlm, 2024-06-07:
> ![image](https://private-user-images.githubusercontent.com/24932258/337557929-3d0ac03e-5003-429a-b191-a74d77fb498d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Could you please first copy the model and necessary files to a separate directory and run 'gz sim -r'?
(Don't forget to export GZ_SIM_RESOURCE_PATH)

xdwgood, 2024-06-07:
[image]

serkanMzlm, 2024-06-07:
@xdwgood 
Could you please follow the steps listed below in order?
```
# 1. Download my github page
git clone git@github.com:serkanMzlm/gz-sim.git
cd gz-sim/

# 2. set GZ_SIM_RESOURCE_PATH location
export GZ_SIM_RESOURCE_PATH=/home/${USER}/gz-sim/examples

# 3. run world
gz sim -r examples/worlds/mountain_fog.sdf 

```

[image]

serkanMzlm, 2024-06-07:
> ![Screenshot from 2024-06-07 16-06-15](https://private-user-images.githubusercontent.com/24932258/337578788-e033596b-f2c1-487a-bef9-9e8b18f829e7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc3NDk0MDEsIm5iZiI6MTcxNzc0OTEwMSwicGF0aCI6Ii8yNDkzMjI1OC8zMzc1Nzg3ODgtZTAzMzU5NmItZjJjMS00ODdhLWJlZjktOWU4YjE4ZjgyOWU3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA3VDA4MzE0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkMmJiNTkzYmI5NzJjOTdjNzUxZWNmYmNmODA2Njk4MzJkZGYwNzVlM2EyY2ExNWE3ZjljMzhiZGE2NjMyOTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QQCJMWOhqsCzJOeEiyUgCKOQh3bVGuy-wkegArmdT6w)

You need to install the requirements to run DEM files. `sudo apt-get install gdal-bin libgdal-dev python3-gdal`

xdwgood, 2024-06-07:
[image]

@serkanMzlm  Thank you for your help. I'm new to gazebo. I don't know if it is necessary to deal with the err
```
[Err] [SystemPaths.cc:534] File [/worlds/mountain_fog.sdf] resolved to path [/worlds/mountain_fog.sdf] but the path does not exist
[Err] [Server.cc:109] Failed to find world [/worlds/mountain_fog.sdf]）
```

xdwgood, 2024-06-07:
Could we close this question now?

serkanMzlm, 2024-06-07:
@xdwgood No, we cannot close this question now. The issue has not been resolved yet, particularly with the PX4 commands not working

serkanMzlm, 2024-06-07:
> ![image](https://private-user-images.githubusercontent.com/24932258/337596094-b37ff941-87eb-4ccc-9719-9fcb14f16cf8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

You don't have to use only the specified data. You can directly add a black and white image of any map to obtain this map :)

xdwgood, 2024-06-11:
> No, we cannot close this question now. The issue has not been resolved yet, particularly with the PX4 commands not working

I use this model to test the takeoff command, and the drone can take off, as shown in the following figure:
[image]

But then I will encounter the following error, and then gz will stop working.

A: 
0.9243 2547 -501.2 0.09919 2254 10.76 1.066 2255 10.71 -0.1081 2255 10.65 1.273 2254 10.82 -0.09417 2547 -501.4 0.0631 2546 -501.2 0.1326 2547 -501.5 0.2493 2546 -501.2 0.4691 2546 -501.1 -1.31 -1.31 -1.31 -1.31 0 0 
2547 1.577e+07 1.706e+06 -2919 1.466e+07 3.642e+06 4624 1.467e+07 3.643e+06 -1882 1.467e+07 3.644e+06 3587 1.466e+07 3.641e+06 -3184 1.577e+07 1.706e+06 -3970 1.577e+07 1.705e+06 -1756 1.578e+07 1.706e+06 -2340 1.577e+07 1.705e+06 12.6 1.577e+07 1.705e+06 -4955 -4955 -4955 -4955 0 0 
-501.2 1.706e+06 1.431e+07 -2137 3.653e+06 1.089e+07 3249 3.655e+06 1.09e+07 4793 3.656e+06 1.09e+07 -3681 3.651e+06 1.089e+07 574.9 1.707e+06 1.432e+07 -4682 1.705e+06 1.431e+07 798.8 1.707e+06 1.432e+07 -3104 1.705e+06 1.431e+07 61.13 1.706e+06 1.431e+07 8743 8743 8743 8743 0 0 
0.09919 -2919 -2137 1.245 -2980 -2027 -0.7396 -2982 -2028 0.1748 -2982 -2029 0.3309 -2980 -2027 0.9411 -2919 -2137 1.753 -2918 -2136 0.6679 -2920 -2137 1.271 -2918 -2136 0.4604 -2918 -2136 -0.2784 -0.2784 -0.2784 -0.2784 0 0 
2254 1.466e+07 3.653e+06 -2980 1.886e+09 -3.28e+09 4701 1.467e+07 3.649e+06 -1019 1.467e+07 3.648e+06 2740 1.466e+07 3.646e+06 -2825 1.466e+07 3.651e+06 -4313 1.466e+07 3.65e+06 -1488 1.467e+07 3.652e+06 -2592 1.466e+07 3.65e+06 20.45 1.466e+07 3.651e+06 13.23 13.23 13.23 8.404e+06 0 0 
10.76 3.642e+06 1.089e+07 -2027 -3.28e+09 5.773e+09 3110 3.649e+06 1.089e+07 3279 3.651e+06 1.089e+07 -2196 3.643e+06 1.088e+07 -51.9 3.643e+06 1.09e+07 -4077 3.638e+06 1.089e+07 329.4 3.643e+06 1.09e+07 -2659 3.639e+06 1.089e+07 47.32 3.641e+06 1.089e+07 23.09 23.09 23.09 -1.475e+07 0 0 
1.066 4624 3249 -0.7396 4701 3110 2.361 4703 3112 0.8905 4704 3112 0.7306 4700 3110 -0.2842 4624 3249 -1.521 4622 3247 0.148 4625 3249 -0.7701 4623 3248 0.4857 4622 3247 0.3516 0.3516 0.3516 0.3516 0 0 
2255 1.467e+07 3.655e+06 -2982 1.467e+07 3.649e+06 4703 1.886e+09 -3.28e+09 -1020 1.468e+07 3.652e+06 2741 1.467e+07 3.651e+06 -2827 1.467e+07 3.653e+06 -4315 1.467e+07 3.652e+06 -1488 1.467e+07 3.654e+06 -2593 1.467e+07 3.652e+06 20.46 1.467e+07 3.653e+06 7.28 7.28 8.404e+06 7.28 0 0 
10.71 3.643e+06 1.09e+07 -2028 3.649e+06 1.089e+07 3112 -3.28e+09 5.774e+09 3280 3.655e+06 1.089e+07 -2197 3.647e+06 1.088e+07 -51.86 3.644e+06 1.09e+07 -4079 3.64e+06 1.09e+07 329.7 3.645e+06 1.09e+07 -2660 3.641e+06 1.09e+07 47.35 3.642e+06 1.09e+07 32.28 32.28 -1.475e+07 32.28 0 0 
-0.1081 -1882 4793 0.1748 -1019 3279 0.8905 -1020 3280 2.462 -1019 3281 -1.396 -1020 3278 1.175 -1882 4794 -0.5596 -1882 4791 1.029 -1882 4794 -0.2585 -1882 4792 0.49 -1881 4791 3.872 3.872 3.872 3.872 0 0 
2255 1.467e+07 3.656e+06 -2982 1.467e+07 3.651e+06 4704 1.468e+07 3.655e+06 -1019 1.888e+09 -3.283e+09 2741 1.467e+07 3.653e+06 -2827 1.467e+07 3.655e+06 -4316 1.467e+07 3.653e+06 -1488 1.468e+07 3.656e+06 -2594 1.467e+07 3.654e+06 20.47 1.467e+07 3.654e+06 7.09 8.408e+06 7.09 7.09 0 0 
10.65 3.644e+06 1.09e+07 -2029 3.648e+06 1.089e+07 3112 3.652e+06 1.089e+07 3281 -3.283e+09 5.779e+09 -2198 3.646e+06 1.089e+07 -51.79 3.645e+06 1.091e+07 -4080 3.64e+06 1.09e+07 329.8 3.645e+06 1.091e+07 -2661 3.641e+06 1.09e+07 47.36 3.643e+06 1.09e+07 25.38 -1.475e+07 25.38 25.38 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
A(10, 11): -3.283e+09
A(11, 10): -3.283e+09
gz sim --verbose=1 -r -s /home/xdw/PX4-Autopilot/Tools/simulation/gz/worlds/mountain_fog.sdf: ./dart/constraint/BoxedLcpConstraintSolver.cpp:229: virtual void dart::constraint::BoxedLcpConstraintSolver::solveConstrainedGroup(dart::constraint::ConstrainedGroup&): Assertion `isSymmetric( n, mA.data(), mOffset[i], mOffset[i] + constraint->getDimension() - 1)' failed.
[... cut ...]
#5    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x752a0e9442ca, in dart::constraint::BoxedLcpConstraintSolver::solveConstrainedGroup(dart::constraint::ConstrainedGroup&)
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x752a51639e95, in __assert_fail
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x752a5162871a, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x752a516287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x752a51642475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x752a516969fc, in pthread_kill
Aborted (Signal sent by tkill() 14653 1000)

@serkanMzlm

xdwgood, 2024-06-11:
Maybe the drone hit the mountain, which caused this problem?
[image]

serkanMzlm, 2024-06-11:
@xdwgood It doesn't get to this part. For some reason, it doesn't load the model; it loads the world environment but doesn't add the model part. 

```bash
make px4_sitl gz_x500_mountain_fog
```

@xdwgood, is the collision part enabled for you?

serkanMzlm, 2024-06-11:
I found the error. When sending the PX4-gazebo-models model, I corrected the `<world name="mountain_fog">` part, but it remained as default when used, and the issue was resolved

closed 2024-06-11 by serkanMzlm