PX4/PX4-Autopilot #23241: Unable to ros2 topic echo /fmu/out/vehicle_local_position and /fmu/out/vehicle_status[Bug]

MonkeScripts, 2024-06-10:
### Describe the bug

I have setup up Micro XRCE-DDS Agent & Client following the[ ROS2 user guide](https://docs.px4.io/main/en/ros/ros2_comm.html#ros-2-subscriber-qos-settings)
****

My flight controller has no gps and is only connected to my Nvidia Jetson Orin Nx Companion computer via ethernet 
[image]

I believed that I have setup the communication bridge correctly between my companion computer and my flight controller because I am able to see the topics being published on the client:
[image]
[image]

On the companion computer, I am able to `ros2 topic list` to show all the topics
[image]

However, some of the ros2 topics can be echo'ed such as `/fmu/out/vehicle_odometry` but some topics, namely `/fmu/out/vehicle_local_position` and `/fmu/out/vehicle_status` cannot be echoed at all 
[image]

I created my own subscriber node to subscribe to `/fmu/out/vehicle_status` but the callback was not triggered

Also, I was able to see the bandwith for `/fmu/out/vehicle_status` 
[image]

**I dont believe its a px4_msgs version mismatch because I cloned from https://github.com/PX4/px4_msgs and the flight controller firmware is from the latest release 1.14.3**
****

### To Reproduce

1. Start XRCE-DDS client on mavlink console
`microxrce_dds_client start`
2. Start XRCE-DDS agent on companion computer 
`MicroXRCEAgent udp4 -p 8888`
3. `ros2 topic list` to list all topics
4. `ros2 topic echo /fmu/out/vehicle_status` and check for topic 

### Expected behavior

Expected to see the topic data from terminal

### Flight Log

No flight log

### Software Version

HW arch: PX4_FMU_V6X
HW type: V6X
HW FMUM ID: 0x003
HW BASE ID: 0x000
PX4 git-hash: 1dacb4cdef2d7145754fc788fa8dc482eed74b40
PX4 version: Release 1.14.3 (17695743)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: May 30 2024 03:33:40
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000303736303532510d00480029
MCU: STM32H7[4|5]xxx, rev. V

px4_msgs version: 
https://github.com/PX4/px4_msgs

### Flight controller

Pixhawk6x

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

DEVICE=eth0
BOOTPROTO=fallback
NETMASK=255.255.255.0
IPADDR=192.168.1.5
ROUTER=192.168.0.254
DNS=192.168.0.254

Companion computer netplan:
network:
  version: 2
  renderer: NetworkManager
  ethernets:
      eth0:
          addresses:
              - 192.168.1.45/24
          nameservers:
              addresses: [8.8.8.8, 8.8.4.4]
          #routes:
              #- to: 192.168.0.1
                #via: 192.168.0.1

xdwgood, 2024-06-11:
[image]

I use the latest main branch and 6x flight control, and it works normally.

MonkeScripts, 2024-06-11:
Hi @xdwgood, the main branch refers to the 'main' in https://github.com/PX4/px4_msgs? For the flight controller firmware does it belong under this tag(https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.3)?

MonkeScripts, 2024-06-11:
Hi, since the message definitions keep on changing is it better to revert to the stable v1.14.0 and how do you upload a custom firmware to the board?