PX4/PX4-Autopilot #23251: [Bug] The mavlink module has heap-overflow

zhangteng0526, 2024-06-11:
### Describe the bug

When sending a mavlink message to the drone, ASAN detected a heap-overflow and caused the software to exit.

### To Reproduce

1. Start the drone software simulation
2. Send a mavlink message to the drone, msgid is 332, and the waypoint field is greater than 5
3. The software-in-the-loop simulation crashes

### Expected behavior

The software-in-the-loop simulation crashes

### Screenshot / Media

[image]

### Flight Log

。

### Software Version

v1.12.3

### Flight controller

,

### Vehicle type

None

### How are the different components wired up (including port information)

,

### Additional context

,

federicociresola, 2024-06-11:
@zhangteng0526 can you reload the image? because it has loaded wrong

dagar, 2024-06-12:
If you could run latest PX4 on address sanitizer on current PX4 (latest stable or main branch) that would be helpful. Unfortunately we're not able to maintain old releases (v1.12 was release July 2021).

closed 2024-06-12 by dagar