PX4/PX4-Autopilot #23259: [Bug] Again Stale message error when the gz is running

federicociresola, 2024-06-12:
### Describe the bug

When I run the gazebo I obtain from px4 the following error:
```
...
ERROR [sensors] Accel #0 fail:  STALE!
ERROR [sensors] Gyro #0 fail:  STALE!
...
```
I thought that has been resolved with the #22767 

### To Reproduce

1. clone the PX4 repository
2. `make px4_sitl gz_x500`

### Expected behavior

px4 without error during the simulation

### Screenshot / Media

[image]

### Flight Log

- 

### Software Version

- ROS2 version: **Humble**
- Gazebo version: **Garden**
- OS version: **Ubuntu 22.04**
- PX4 version: **last main**

### Flight controller

x500

### Vehicle type

Multicopter

PerFrivik, 2024-06-12:
Is this a freshly cloned vanilla PX4, also did you try a `git submodule update --recursive --init` ?

federicociresola, 2024-06-17:
I have tried, but I still have the problem

StefanoColli, 2024-06-18:
How did you install Gazebo? From source or by using the ubuntu.sh script in the px4 repo?

federicociresola, 2024-06-18:
I installed it from ROS2 package with the following command
```
sudo apt install ros-humble-ros-gzgarden
```
Is it possible that this is the problem?

federicociresola, 2024-06-18:
I tried to remove the Gazebo garden installed through ROS2 and I installed gz-garden but there is still the problem
```
# Remove ros2 gz garden
sudo apt remove ros-humble-ros-gzgarden ros-humble-ros-gzgarden-sim

# Install gz garden from PX4 shell script
./Tools/setup/ubuntu.sh
```

PerFrivik, 2024-06-18:
Can you install gazebo through the PX4 installation guide please? There may be conflicting versions issues, as PX4 does not use the newest GZ version. 

https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html#simulation-and-nuttx-pixhawk-targets 

Please try setting up the dev environment and see if that helps.

federicociresola, 2024-06-19:
you were right there was a conflict in somewhere with Gazebo version of the package manager of Ubuntu and ROS2 Humble.

Thank you all for your help!

I leave here the brute force procedure that I have used (be careful when using it):
```
# ROS2
sudo apt remove --purge ros-humble-ros-gzgarden*

# Ubuntu apt gazebo
sudo dpkg --list | grep gazebo
sudo apt remove --purge  *gazebo*

# Ubuntu apt ignition
sudo dpkg --list | grep  ignition
sudo apt remove --purge *ignition*

sudo apt autoremove
```

closed 2024-06-19 by federicociresola