PX4/PX4-Autopilot #23281: Add feature for CRSF RX binding via cli

benjinne, 2024-06-17:
### Describe problem solved by the proposed feature

Add ability to bind CRSF receivers from flight controller

Example code was introduced in betaflight:
https://github.com/betaflight/betaflight/pull/13119

### Describe your preferred solution

Add CLI or UI method to bind CRSF receivers

### Describe possible alternatives

n/a

dagar, 2024-06-17:
Is this officially documented anywhere?

benjinne, 2024-06-17:
> Is this officially documented anywhere?

Sort-of in the ELRS documentation: 
https://www.expresslrs.org/quick-start/binding/#updated-binding-procedure-since-expresslrs-340

benjinne, 2024-06-18:
I've opened a PR but don't have hardware to test. If anyone want's to give a try and verify it works, we can merge it in

benjinne, 2024-08-21:
I've verified on my hardware that my PR works. We can close this once it's merged

closed 2026-03-18 by github-actions[bot]