PX4/PX4-Autopilot #23305: TF-mini i2c port support

dfordequan, 2024-06-20:
### Describe problem solved by the proposed feature

TF mini is a nice range finder sensor that works in outdoors, and it supports both UART and I2C interference. However, PX4 can only work with UART port with TFmini. The I2C driver is missing. That's a pity for controllers like 6c mini that doesn't gave many serial ports.

### Describe your preferred solution

Install the TFmini i2c driver in the firmware.

### Describe possible alternatives

Not sure

dagar, 2024-06-20:
I don't have one of this modules, but if anyone has access and is interested I could help you get the driver structure in place fairly quickly.

justas-, 2024-07-30:
I've done it before (years back). You need first to use the config utility to reconfigure the TF-Mini to use I2C mode. A word of advice - TF Mini is notoriously noisy (as in electrical and RF noise).