PX4/PX4-Autopilot #23308: [Bug] Distance sensor LIDAR Lightware SF45 Serial not working on fmu-v6x

brunopedro1, 2024-06-21:
### Describe the bug

The distance sensor LIDAR Lightware SF45 Serial is not publishing to the distance_sensor uORB topic on fmu-v6x using a Pixhawk 6X on the Pixhawk RPI CM4 Baseboard. 

First, I started the driver in TELEM3 serial port (or configured the parameter SENS_EN_SF45_CFG to TELEM3):
```
nsh> lightware_sf45_serial start -d /dev/ttyS1
``` 
The sensor starts rotating, as supposed, but no distance measurements are read or published in the distance_sensor uORB topic:
```
nsh> ightware_sf45_serial status
lightware_sf45_serial: read: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
lightware_sf45_serial: com_err: 0 events
```
```
nsh> listener distance_sensor
never published
```

There is also a weird behaviour when I stop the driver and start it again in the same serial port:
```
nsh> lightware_sf45_serial stop
nsh> lightware_sf45_serial start -d /dev/ttyS1
ERROR [px4_work_queue] wq:ttyS1 not available
ERROR [px4_work_queue] init failed
```

(It works well on fmu-v3 using a Hex Cube Black Flight Controller)

### To Reproduce

1. Enable lightware_sf45_serial driver on boardconfig
2. make px4_fmu-v6x_default
3. Start lightware_sf45_serial driver on the desired serial port, for example, using MAVLink Console in QGC
4. Listen to distance_sensor uORB topic, for example, using MAVLink Console in QGC

### Expected behavior

The lightware_sf45_serial driver was expected to publish the measurements of the sensor in the distance_sensor uORB topic.

It works well on fmu-v3 using a Hex Cube Black Flight Controller.

### Flight Log

/

### Software Version

1.14.3

### Flight controller

Pixhawk 6X on Pixhawk RPI CM4 Baseboard

### Vehicle type

None

### How are the different components wired up (including port information)

Pixhawk RPI CM4 Baseboard is powered through the PM03D Power Module with a 2S Li-Po Battery and connected to a computer (GCS) through USB-C cable.   

LiDAR Lightware SF45 is powered throught 5V BEC of the PM03D and connected to TELEM3 serial power of the pixhawk (with common ground between TELEM3 port and power)

dirksavage88, 2024-06-21:
Getting the mavlink status is just confirming that mavlink isn't running on the serial port you want to run the SF45 driver on. You will need to match the baud rate of 921600 on whichever telem port you use. 57600 won't work

brunopedro1, 2024-06-21:
> Getting the mavlink status is just confirming that mavlink isn't running on the serial port you want to run the SF45 driver on. You will need to match the baud rate of 921600 on whichever telem port you use. 57600 won't work

Changed the default baudrate of TELEM3 to 921600 in https://github.com/PX4/PX4-Autopilot/blob/30ce560e3a406fd0084ae52e6ce0d9ef36f4cd3d/Tools/serial/generate_config.py#L97

And, also, included the line 
```
param set-default SER_TEL3_BAUD 921600
``` 
in the initialization script file for standard plane airframe https://github.com/PX4/PX4-Autopilot/blob/30ce560e3a406fd0084ae52e6ce0d9ef36f4cd3d/ROMFS/px4fmu_common/init.d/airframes/2100_standard_plane#L4

Such that the  SER_TEL3_BAUD parameter appears in QGC with 921600: 

```
nsh> param show SER_TEL3_BAUD
Symbols: x = used, + = saved, * = unsaved
x   SER_TEL3_BAUD [1060,1436] : 921600

 1200/1866 parameters used.
``` 

But it is still not publishing in the distance_sensor topic.

dirksavage88, 2024-06-21:
Can you print the driver status? I don't have a 6x to test on, but sounds like it's getting data but unsure if it's the right data. If it worked on a previous autopilot than I am really not sure where the issue could be other than the serial port you are using?

dirksavage88, 2024-06-22:
@brunopedro1  you need a couple of settings preconfigured in [lightware studio](https://lightwarelidar.com/resources-software/)

Connect to a computer over usb and set these settings using lightware studio

The first one is scan on startup:
[image]

Next is baudrate, which must be set to 921600:
[image]

brunopedro1, 2024-06-23:
My configuration in Lightware Studio of the sensor already had the scan on startup and baudrate of 921600. 

I have now tried using the 1.14.2 PX4 release instead of 1.14.3 and, fortunately, the distance_sensor topic is aleady published (connected in TELEM3). However, there are errors in the measurements of current_distance:

```
nsh> listener distance_sensor

TOPIC: distance_sensor
 distance_sensor
    timestamp: 123153891 (0.014098 seconds ago)
    device_id: 7536653 (Type: 0x73, SERIAL:1 (0x00))
    min_distance: 0.20000
    max_distance: 50.00000
    current_distance: -1.00000
    variance: 0.00000
    h_fov: 0.00000
    v_fov: 0.00000
    q: [0.00000, 0.00000, 0.00000, 0.00000] (Roll: 0.0 deg, Pitch: -0.0 deg, Yaw: 0.0 deg)
    signal_quality: 0
    type: 0
    orientation: 0
``` 

In this case, this is the driver status:
```
nsh> lightware_sf45_serial status
lightware_sf45_serial: read: 2652 events, 25928659us elapsed, 9777.02us avg, min 120us max 21092us 1603.900us rms
lightware_sf45_serial: com_err: 0 events
```

dirksavage88, 2024-06-23:
> My configuration in Lightware Studio of the sensor already had the scan on startup and baudrate of 921600.
> 
> I have now tried using the 1.14.2 PX4 release instead of 1.14.3 and, fortunately, the distance_sensor topic is aleady published (connected in TELEM3). However, there are errors in the measurements of current_distance:
> 
> ```
> nsh> listener distance_sensor
> 
> TOPIC: distance_sensor
>  distance_sensor
>     timestamp: 123153891 (0.014098 seconds ago)
>     device_id: 7536653 (Type: 0x73, SERIAL:1 (0x00))
>     min_distance: 0.20000
>     max_distance: 50.00000
>     current_distance: -1.00000
>     variance: 0.00000
>     h_fov: 0.00000
>     v_fov: 0.00000
>     q: [0.00000, 0.00000, 0.00000, 0.00000] (Roll: 0.0 deg, Pitch: -0.0 deg, Yaw: 0.0 deg)
>     signal_quality: 0
>     type: 0
>     orientation: 0
> ```
> 
> Not all measurements present error, but most of them do.
> 
> Using TELEM1 instead (after changing its baudrate to 921600) presents similar results, but the number of measurements with error seems to be lower.
> 
> In this case, this is the driver status:
> 
> ```
> nsh> lightware_sf45_serial status
> lightware_sf45_serial: read: 2652 events, 25928659us elapsed, 9777.02us avg, min 120us max 21092us 1603.900us rms
> lightware_sf45_serial: com_err: 0 events
> ```
> 
> The errors in the measurements are not related to the distance of the sensor to any obstacles being lower than the minimum or higher than the maximum, because it is pointed to a reasonable area, in which the sensor gives accurate results in fmu-v3.

At this point is probably necessary to attach to the nuttx console via the debug port on the 6x, since there isn't much else you can do through mavlink console on QGC for debugging.

dirksavage88, 2024-06-26:
It looks like TELEM3 is on USART2 which  has a baud rate of 57600 in the nuttx config, and dma disabled. Attaching to this port did not work out of the box (probably possible with some nuttx kconfig changes)

[image]

TELEM2 worked out of the box for me without configuring anything other than enabling the sensor in the boardconfig and setting it to TELEM2 on the 6x. TELEM2 is on uart5, which has larger buffer sizes, 115200 baud, and dma enabled

[image]

brunopedro1, 2024-06-27:
> It looks like TELEM3 is on USART2 which has a baud rate of 57600 in the nuttx config, and dma disabled. Attaching to this port did not work out of the box (probably possible with some nuttx kconfig changes)
> 
> ![6x_usart2](https://private-user-images.githubusercontent.com/35986980/343125863-83fd6bf7-e024-49b7-bbeb-19c11229d589.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NzgwNDcsIm5iZiI6MTcxOTQ3Nzc0NywicGF0aCI6Ii8zNTk4Njk4MC8zNDMxMjU4NjMtODNmZDZiZjctZTAyNC00OWI3LWJiZWItMTljMTEyMjlkNTg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI3VDA4NDIyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMwOTU1YmZiYmEwMmU1ODI2YjZjNzBlZWY4OTFjNjM3OTgwNzA5NDdiMjA1MGNkZjAyOWUyMmU3Y2YxNDUzMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CyGzGuugEcEwzT5XUmRZlkBAgTpx8pJc8iSivACj4B8)
> 
> TELEM2 worked out of the box for me without configuring anything other than enabling the sensor in the boardconfig and setting it to TELEM2 on the 6x. TELEM2 is on uart5, which has larger buffer sizes, 115200 baud, and dma enabled
> 
> ![6x_uart5](https://private-user-images.githubusercontent.com/35986980/343125945-3bb95d85-c13c-460d-aae4-31e40ce684a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NzgwNDcsIm5iZiI6MTcxOTQ3Nzc0NywicGF0aCI6Ii8zNTk4Njk4MC8zNDMxMjU5NDUtM2JiOTVkODUtYzEzYy00NjBkLWFhZTQtMzFlNDBjZTY4NGE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI3VDA4NDIyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4NGI5NThiYTA3YWM5MzEzMWQyMTExMjdmMDNjNWYyMGE5NmU1N2I3NmQwZWY1NDEzMzg3ZDlhMjdlZWI0NWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.T6eQ7ElxV2aA3dQV6XoVenXfXotcoxi64NW95LuNGkY)
> 
> TEL2 is the default in the module [yaml](https://github.com/dirksavage88/Firmware/blob/30ce560e3a406fd0084ae52e6ce0d9ef36f4cd3d/src/drivers/distance_sensor/lightware_sf45_serial/module.yaml#L7) file since not all autopilots have 3 telem ports (6c mini for example)

In the Pixhawk RPi CM4 Baseboard, TELEM2 is internally connected to the CM4 so I cannot use it for the LiDAR, unfortunately. 

I tried changing the USART2 nuttX config, increasing the transmit buffer size to 10000 and the baudrate to 115200 (or 912600), but it doesn't seem to make any difference. Also enabled DMA support for USART2 in the config, but it may require a few more steps, because the error "USART2 DMA map not defined (DMAMAP_USART2_RX)" appears when compiling. Despite this, I don't think DMA is the source of the issue, since there are also problems using TELEM1, which is on UART7 and has DMA enabled. Increasing transmit buffer size and baudrate of UART7 and connecting the LiDAR to TELEM1 didn’t make any difference either.

This bug is very strange, since LIDAR works fine on other pixhawks. Everything points to a configuration problem specific to this board.

I really appreciate your help so far in trying to resolve this issue.

dirksavage88, 2024-06-27:
@brunopedro1 I wouldn't rule out the physical connection itself. As mentioned earlier if you get some data on TELEM3 but not valid distance data, your best bet is to use the debug port. There are debug messages that the SF45 driver prints out

Claudio-Chies, 2024-08-06:
@dirksavage88 
were you ever able to find out more regarding this issue?
i have a similar issue, SF45 on TELEM1, powered externaly.
Lightware configured as above.
TELEM1 Baudrate to 921600
Hardware: v5x
but i don't get anything when looking at the status, and if i do a dmesg i see `start of packet not valid` and `CRC mismatch`

`distance_sensor` or `obstacle_distance` also never gets published
```
nsh> lightware_sf45_serial status
lightware_sf45_serial: read: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
lightware_sf45_serial: com_err: 0 events
```

dmesg:
```
INFO  [lightware_sf45_serial] INFO: Product name
INFO  [lightware_sf45_serial] INFO: Send byte: 170
INFO  [lightware_sf45_serial] INFO: Send byte: 64
INFO  [lightware_sf45_serial] INFO: Send byte: 0
INFO  [lightware_sf45_serial] INFO: Send byte: 0
INFO  [lightware_sf45_serial] INFO: Send byte: 112
INFO  [lightware_sf45_serial] INFO: Send byte: 159
INFO  [lightware_sf45_serial] INFO: Product name
INFO  [lightware_sf45_serial] INFO: Send byte: 170
INFO  [lightware_sf45_serial] INFO: Send byte: 64
INFO  [lightware_sf45_serial] INFO: Send byte: 0
INFO  [lightware_sf45_serial] INFO: Send byte: 0
INFO  [lightware_sf45_serial] INFO: Send byte: 112
INFO  [lightware_sf45_serial] INFO: Send byte: 159
INFO  [lightware_sf45_serial] INFO: CRC mismatch
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: payload length invalid, restarting data request
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: CRC mismatch
INFO  [lightware_sf45_serial] INFO: start of packet not valid
nsh> INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: payload length invalid, restarting data request
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
INFO  [lightware_sf45_serial] INFO: start of packet not valid
```

dirksavage88, 2024-08-06:
> @dirksavage88 were you ever able to find out more regarding this issue? i have a similar issue, SF45 on TELEM1, powered externaly. Lightware configured as above. TELEM1 Baudrate to 921600 Hardware: [...]

No, I haven't figured out what the issues are with specific uart ports. I did find that on a recent version of a 6x, TELEM3 did not work but TELEM 2 did. 

The OP mentions TELEM1 on the 6x also not working, but is DMA enabled. So likely not a DMA issue unless there is a flow control + DMA issue.

For the px4 v5x it doesn't look like you have DMA on TELEM1: https://github.com/PX4/PX4-Autopilot/blob/1dacb4cdef2d7145754fc788fa8dc482eed74b40/boards/px4/fmu-v5x/nuttx-config/include/board_dma_map.h#L85

I have seen CRC mismatch errors with long uart cable lengths, but usually I can at least get the start of packet correctly, and the driver starts and reports data with maybe a few CRC errors here and there (it throws out those packets in the driver)

dirksavage88, 2024-08-15:
@Claudio-Chies I believe the issue is related to sending data to the sensor to get rangefinder measurements started. We get some data back but it's not a continuous stream, so the sensor stops sending data after that initial burst. I tried on TEL1, TEL2, & TEL3 and the only issue I had was with TEL3 where it fails to start on the intial try, but after rebooting the  6x it picks it up and reports distance data.

I have attached a patch for the holybro 6x, you may need to modify it. It will now report comms errors if we have issues receiving and sending data. plus flush out any unread data after an error.

[sf45.patch.gz](https://github.com/user-attachments/files/16625423/sf45.patch.gz)

closed 2024-09-23 by alexcekay