PX4/PX4-Autopilot #23318: [Bug] Failed to create threads for work queues on PX4 SITL with Gazebo

pjanuszewski, 2024-06-22:
### Describe the bug

I'm trying to implement a function that calculates control feedback matrix for LQR during runtime, in the mc_att_control module. The functions used to calculate the matrix are from ct-toolbox library: [https://ethz-adrl.github.io/ct/ct_doc/doc/html/index.html](url). The `lqrtest()` function for verifying how the LQR matrix is being computed, was added to `init()` function of `MulticopterAttitudeControl` class. The project is compiling without errors with `make`, however when I try to start a SITL simulation with `make px4_sitl gz_x500`, I keep getting errors about failure to create threads:

```
user@laptop:~/Git_repos/PX4-Autopilot$ make px4_sitl gz_x500
[0/1] cd /home/user/Git_repos/PX4-Autopilot/build/px4_sitl_default/src/..._x500 /home/user/Git_repos/PX4-Autopilot/build/px4_sitl_default/bin/px4
ERROR [px4_work_queue] Thread ID: 127847736363290
ERROR [px4_work_queue] failed to create thread for wq:lp_default (22): Invalid argument
ERROR [px4_work_queue] Error occurred at /home/user/Git_repos/PX4-Autopilot/platforms/common/px4_work_queERROR [px4_work_queue] Thread attributes: stack size = 61440, priority = 66
ERROR [px4_work_queue] failed to create wq:lp_default
ERROR [px4_work_queue] wq:lp_default not available
ERROR [px4_work_queue] init failed

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4001
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 296 bytes, decoded 296 bytes (INT32:12, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/user/Git_repos/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO  [gz_bridge] world: default, model name: x500_0, simulation model: x500
ERROR [px4_work_queue] Thread ID: 127847736363290
ERROR [px4_work_queue] failed to create thread for wq:rate_ctrl (22): Invalid argument
ERROR [px4_work_queue] Error occurred at /home/user/Git_repos/PX4-Autopilot/platforms/common/px4_work_queERROR [px4_work_queue] Thread attributes: stack size = 61440, priority = 99
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
ERROR [px4] Startup script returned with return value: 2

```

I have tried to increase stack size of the problematic threads, but it doesn't help.

### To Reproduce

All the files that have been changed / added are listed below. They should all be in `PX4-Autopilot/src/modules/mc_att_control` directory. The file type has to be changed to .hpp in files 2-6, and to .cpp in file 7. After adding them, build project with cmake and start the simulation with `make px4_sitl gz_x500`.

1. [CMakeLists.txt](https://github.com/user-attachments/files/15937601/CMakeLists.txt)
2. [LQR-impl.txt](https://github.com/user-attachments/files/15937923/LQR-impl.txt)
3. [CARE.txt](https://github.com/user-attachments/files/15937603/CARE.txt)
4. [CARE-impl.txt](https://github.com/user-attachments/files/15937604/CARE-impl.txt)
5. [LQR.txt](https://github.com/user-attachments/files/15937606/LQR.txt)
6. [mc_att_control.txt](https://github.com/user-attachments/files/15937607/mc_att_control.txt)
7. [mc_att_control_main.txt](https://github.com/user-attachments/files/15937944/mc_att_control_main.txt)

### Expected behavior

During the startup of the simulation, the K matrix should be calculated and written to a file in `PX4Autopilot/src/modules/mc_att_control` directory.

### Flight Log

N/A

### Software Version

**Environment:**
- OS: Ubuntu 22.04.4
- PX4 version: v1.14.3
- Build command: `make px4_sitl gz_x500`
- Compiler: gcc 11.4.0

### Flight controller

N/A

### Vehicle type

Multicopter

xdwgood, 2024-06-24:
[image]

I encountered this error when I tried to compile. Do you know how to solve it? thank you

pjanuszewski, 2024-06-24:
@xdwgood I think some necessary libraries are not installed on your system, or are not correctly detected by CMake.
Firstly, you could make sure that they are properly installed. You can run:
```
sudo apt-get install libpthread-stubs0-dev
sudo apt-get install libblas-dev
sudo apt-get install liblapack-dev
```
After you install them, they also have to be detected and linked by CMake. The  
> 1. [CMakeLists.txt](https://github.com/user-attachments/files/15944153/CMakeLists.txt)

file from my issue should already have the commands that link the `LAPACK` library to the module. Specifically the:
```
find_package(LAPACK REQUIRED)
```
```
target_compile_definitions(modules__mc_att_control PRIVATE CT_USE_LAPACK)

if (LAPACK_FOUND)
	include_directories(${LAPACK_INCLUDE_DIRS})
	target_link_libraries(modules__mc_att_control PUBLIC ${LAPACK_LIBRARIES})
else()
	message(FATAL_ERROR "LAPACK library not found.")
endif()
```
Make sure these parts are in your CMakeLists file. Let me know if this helps.

xdwgood, 2024-06-25:
I have installed those libraries, but I get the same problem.I use the same software version and environment as yours, except for the px4 version (I use the latest main branch).

xdwgood, 2024-06-25:
[image]

pjanuszewski, 2024-06-25:
I'm not sure why it doesn't want to accept the lapack config file. I'm also using the lapack 3.12.0 and it links it normally if you include all the necessary parts in the CMakeLists file.
```set(LAPACK_LIBRARIES /usr/local/lib/liblapack.a)
set(LAPACK_INCLUDE_DIRS /usr/local/include)
set(BLAS_LIBRARIES /usr/local/lib/libblas.a)
set(FORTRAN_LIBRARIES gfortran)
```
```
target_compile_definitions(modules__lqr_module PRIVATE CT_USE_LAPACK)

if (LAPACK_FOUND)
    include_directories(${LAPACK_INCLUDE_DIRS})
    target_link_libraries(modules__lqr_module PUBLIC ${LAPACK_LIBRARIES} ${BLAS_LIBRARIES} ${FORTRAN_LIBRARIES})
else()
    message(FATAL_ERROR "LAPACK library not found.")
endif()
```