PX4/PX4-Autopilot #23340: Invalid mission with RTL crash

ryanjAA, 2024-06-28:
### Describe the bug

On fixed-wing, if you upload a bad mission (fails feasibility) while in the air, it still gets on the vehicle. So if you go into return mode it crashes in varying ways depending on the firmware version. This should be changed and understood.

Tried this on a real-flight and it did crash so here is the SITL follow up.

### To Reproduce

Set RTL type to 2 (planned mission landing)

Take off with a mission planned. Upload a new mission that fails checks while in the air (or just delete the mission from the vehicle), both seem to do the same thing but this should be checked. Then put it in return (with no or a bad mission on the vehicle) and here's what happens:

• 1.14.0 it descends on its way to the home location all the way to 0m alt and crashes. Alt SP gets set to 0. 

• 1.14.3 it climbed to 75m (but i had it set to 50m rtl alt). It then went home and once there it said landing while in a loiter but stayed in return mode and crash landed (didnt go in land mode and i didnt want nor expect it to). Also not ideal.

• v1.15.0-beta2 tag - it worked correctly, said no valid mission, loitering

• Main from today it worked correctly, said no valid mission, loitering

Where was this change? I'd like to backport it to 1.14.3 since it can make a vehicle crash at its most vulnerable point (a return...)

### Expected behavior

Not crash... or at least do what it is doing in main and reject the return.

### Flight Log

If needed, will upload...

### Software Version

Tested on 1.14.0, .3, v1.15.0-beta2 and main

### Flight controller

SITL and Pix 6c 

### Vehicle type

Fixed Wing

sfuhrer, 2024-07-08:
We have done a lot of work since 1.14 was branched off towards more stable (Mission-) RTL functionality. The biggest chuck came in with https://github.com/PX4/PX4-Autopilot/pull/20903, but there were then also a couple of follow ups, so I would be very careful in cherry-picking or then use v1.15.

ryanjAA, 2024-07-08:
Thanks @sfuhrer 

I’ll take a look then. Until 1.15 is out maybe we need a note for new users since still does this on current 1.14.3 release.  Appreciate you sending it across. 

@hamishwillee where would that go for new users? 

Closing as not going to fix but just need to acknowledge is all.

closed 2024-07-08 by ryanjAA