PX4/PX4-Autopilot #23357: The heading angle calculation of dual antenna GPS is incorrect[Bug]

fanchuannangang1503, 2024-07-03:
### Describe the bug

In the handleHeading function in nmea.cpp, there is a statement "heading_rad -= _heading_offset", which is problematic because heading_rad is in the NED coordinate system, while _heading_offset is in the body coordinate system.
In the updateGpsYaw function in gps_yaw_fusion.cpp, there is a statement "measured_hdg = wrap_pi(gps_sample.yaw + _gps_yaw_offset)", which also directly adds gps_sample.yaw in the NED coordinate system to _gps_yaw_offset in the body coordinate system.
Moreover, _heading_offset is subtracted in nmea.cpp, but _gps_yaw_offset is added in gps_yaw_fusion.cpp, which cancels each other out. It seems that the GPS_YAW_OFFSET parameter does not work.

### To Reproduce

xxx

### Expected behavior

xxx

### Flight Log

xxx

### Software Version

V1.14.3

### Flight controller

ALL Flight controller

### Vehicle type

None

wanglehui, 2024-07-09:
Btw， Ekf::resetYawToGps said “GPS yaw measurement is alreday compensated for antenna offset in the driver”
：-<

fanchuannangang1503, 2024-07-09:
> Btw， Ekf::resetYawToGps said “GPS yaw measurement is alreday compensated for antenna offset in the driver” ：-<

However, the calculation of the gnss_yaw parameter of the resetYawToGps function is incorrect. Because nmea.cpp should not subtract _heading_offset in the BODY frame from heading_rad in the NED frame.

fanchuannangang1503, 2024-07-09:
[image]
SensorGps.msg said "[heading] is in ned frame, [heading_offset] is in body frame"

bresch, 2024-07-09:
> In the handleHeading function in nmea.cpp, there is a statement "heading_rad -= _heading_offset", which is problematic because heading_rad is in the NED coordinate system, while _heading_offset is in the body coordinate system.

Right, this is not correct. However it doesn't have a big impact as the EKF [removes this correction](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/aid_sources/gnss/gps_yaw_fusion.cpp#L52) and [fuses it correctly](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/python/ekf_derivation/derivation.py#L569-L592).

> In the updateGpsYaw function in gps_yaw_fusion.cpp, there is a statement "measured_hdg = wrap_pi(gps_sample.yaw + _gps_yaw_offset)", which also directly adds gps_sample.yaw in the NED coordinate system to _gps_yaw_offset in the body coordinate system.

Yes, the [comment is incorrect](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/aid_sources/gnss/gps_yaw_fusion.cpp#L51-L52), but it effectively removes the wrongly applied heading offset from the driver.

> Moreover, _heading_offset is subtracted in nmea.cpp, but _gps_yaw_offset is added in gps_yaw_fusion.cpp, which cancels each other out. It seems that the GPS_YAW_OFFSET parameter does not work.

That's not true. The `GPS_YAW_OFFSET` is used to [predict the measurement of the GNSS heading](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/python/ekf_derivation/derivation.py#L580-L587).

fanchuannangang1503, 2024-07-09:
> > In the handleHeading function in nmea.cpp, there is a statement "heading_rad -= _heading_offset", which is problematic because heading_rad is in the NED coordinate system, while _heading_offset is in the body coordinate system.
> 
> Right, this is not correct. However it doesn't have a big impact as the EKF [removes this correction](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/aid_sources/gnss/gps_yaw_fusion.cpp#L52) and [fuses it correctly](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/python/ekf_derivation/derivation.py#L569-L592).
> 
> > In the updateGpsYaw function in gps_yaw_fusion.cpp, there is a statement "measured_hdg = wrap_pi(gps_sample.yaw + _gps_yaw_offset)", which also directly adds gps_sample.yaw in the NED coordinate system to _gps_yaw_offset in the body coordinate system.
> 
> Yes, the [comment is incorrect](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/aid_sources/gnss/gps_yaw_fusion.cpp#L51-L52), but it effectively removes the wrongly applied heading offset from the driver.
> 
> > Moreover, _heading_offset is subtracted in nmea.cpp, but _gps_yaw_offset is added in gps_yaw_fusion.cpp, which cancels each other out. It seems that the GPS_YAW_OFFSET parameter does not work.
> 
> That's not true. The `GPS_YAW_OFFSET` is used to [predict the measurement of the GNSS heading](https://github.com/PX4/PX4-Autopilot/blob/419652b9fed281ca187b1f24209e4dce9016c7d5/src/modules/ekf2/EKF/python/ekf_derivation/derivation.py#L580-L587).
> 
> In summary we should simply remove the incorrect correction in the driver and also delete the part in ekf2 that removes it.

But Ekf::resetYawToGps() thinks "gnss_yaw" is heading angle of XYZ body frame rel to NED. If the initial attitude of the vehicle is not level with the ground.

bresch, 2024-07-10:
> But Ekf::resetYawToGps() thinks "gnss_yaw" is heading angle of XYZ body frame rel to NED. If the initial attitude of the vehicle is not level with the ground.

True, the reset isn't correct. It can't be really far from the true value as we only reset to GNSS yaw if the tilt is < 30 degrees but we should fix that anyway.

I anyway want to move the heading data out of the GNSS message in the near future; I'll do that at the same time.