PX4/PX4-Autopilot #23379: [Bug] SITL plane does not return to home correctly with RTL_TYPE = 2

anaam-wingxpand, 2024-07-09:
### Describe the bug

If RTL_TYPE = 2 and there is no mission landing waypoint, the SITL plane returns to a random point near Takeoff and tries to land but fails. This ignores the true Home position, and ignores the Return to Launch settings that say to loiter. Furthermore, it fails to land, continuously flying at 0 m altitude.

### To Reproduce

1. $ `make px4_sitl gz_rc_cessna`
2. Upload any mission with a Takeoff but without a Landing.
3. Go to Parameters and change RTL_TYPE to 2 (Return to a planned mission landing, or home)
4. Go to Safety and note the option selected under "Return to launch, then: "
5. Fly the mission, and during the mission hit Return flight mode or trigger a Return failsafe.
6. See error.

### Expected behavior

The SITL plane should either land successfully or wait (depending on the Return to Launch settings) at Home. In my case, Home was the Launch waypoint, where the plane had spawned at. This is the home that the plane would return to with other RTL_TYPEs.

### Screenshot / Media

Note the incorrect home point, and the plane failing to land and continuously flying.

[image]
[image]

### Flight Log

https://review.px4.io/plot_app?log=911929cb-52db-444f-93e6-3c969b94ba5d 

### Software Version

v1.14

### Flight controller

N/A (SITL)

### Vehicle type

None

ryanjAA, 2024-07-10:
This is similar to https://github.com/PX4/PX4-Autopilot/issues/23340

anaam-wingxpand, 2024-07-10:
In the v1.15.0-beta2 release, the SITL plane returns to the correct home position and actually lands with RTL_TYPE = 2. 

However, it still completely ignores the Return to Launch settings. It always lands, and does not change altitude or loiter.

sfuhrer, 2024-07-10:
We won't patch 1.14 anymore, and it's fixed in v1.15.

>ignores the Return to Launch settings.

Which settings? RTL_LAND_DELAY?

anaam-wingxpand, 2024-07-10:
@sfuhrer yes, it ignores RTL_LAND_DELAY and RTL_DESCEND_ALT. It should only ignore those settings if its returning to mission landing, but in this case it is returning to Home.

sfuhrer, 2024-08-14:
> @sfuhrer yes, it ignores RTL_LAND_DELAY and RTL_DESCEND_ALT.

This is actually expected if the RTL destination is a mission landing. The RTL_LAND_DELAY and RTL_DESCEND_ALT parameters are only used for the RTL to Home (without mission landing). 
What's the use case for you, why do you want to configure a delay and descend?

anaam-wingxpand, 2024-08-16:
> This is actually expected if the RTL destination is a mission landing. The RTL_LAND_DELAY and RTL_DESCEND_ALT parameters are only used for the RTL to Home (without mission landing). What's the use [...]

I'm saying that when RTL_TYPE = 2 and it RTLs to Home (without mission landing), it ignores RTL_LAND_DELAY and RTL_DESCEND_ALT. It immediately lands at home. This is unexpected behavior, because as you said, it should be using the delay and descend parameters in this case.

sfuhrer, 2024-08-19:
Ah okay all clear now, and yes, I can reproduce. RTL_LAND_DELAY and RTL_DESCEND_ALT are not implemented in the reverse mission RTL, though I get that you would want them (e.g. for fixed-wing you do not want to land but keep it loitering around Home).
@KonradRudin we'd need to add some more logic around [here](https://github.com/Auterion/PX4_firmware_private/blob/15c4a89ecabebebdb097c359db88ef1347a331ce/src/modules/navigator/rtl_mission_fast_reverse.cpp#L233) I guess. As it for the reverse mission should do the same logic as for the RTL_direct I wonder if we can use a shared library for both.

closed 2026-03-18 by github-actions[bot]